tock/tock #470: SAM4L automatic low power

alevy, 2017-06-27:
* [ ] Put cortex-m4 into deep sleep
  * [ ] Add interface in chip interface to choose sleep mode
  * [ ] Figure out in SAM4L if we're can go into deep sleep mode or not
  * [ ] Do accounting in SAM4L peripherals

alevy, 2017-08-03:
I'm moving this tracking issue to post Tock 1.0, though we're fairly far along. @brghena and @adkinsjd did this fall off the radar, get low priority (seems like Signpost is so far holding up power-wise), or is something blocking testing/reviewing #479?

lschuermann, 2024-10-18:
Is this still something that's worth targeting, especially given the fact that we've pivoted to nRFs as the de-facto reference boards (from SAM4L)?

alevy, 2024-10-20:
I don't think this is even up to date. This was "solved" for Holly Chiang's EWSN '21 paper (https://www.amitlevy.com/papers/2021-ewsn-chiang.pdf). Not all of it was upstreamed as that technique was fairly invasive, but the last time this issue was even measured was well before that paper started exploring the SAM4L again.

The NRF and STM platforms are probably much more interesting in practice: the nRF doing things automatically in hardware if the kernel + apps behave well (which we know we're not necessarily hitting all those marks by default and we should probably figure out what needs doing), and the STM platforms being more similar to the SAM4L in ceding more control over power management to software.

lschuermann, 2024-10-20:
Okay, with that I think it's appropriate to close this issue, especially given its specific to the SAM4L. If anyone wants to tackle this again we can always reopen, or we can create a low-power meta tracking issue.

closed 2024-10-20 by lschuermann