tock/tock #491: `make fmt` in userland error

bradjc, 2017-07-10:
```
$ make fmt
Parsing: adafruit.io.c as language C
Parsing: main.c as language C
/bin/sh: 1: [: unexpected operator
Parsing: ble_http.c as language C
/bin/sh: 1: [: unexpected operator

 $ pwd
/home/bradjc/git/tock/userland/examples/sparkfun-data
```

ppannuto, 2017-07-11:
This is harder for me to test as it works fine on my machine and Travis, but does:

```diff
diff --git a/userland/AppMakefile.mk b/userland/AppMakefile.mk
index 395c4861..c230f66c 100644
--- a/userland/AppMakefile.mk
+++ b/userland/AppMakefile.mk
@@ -208,16 +208,16 @@ fmt format:: $(FORMATTED_FILES)

 $(BUILDDIR)/format/%.uncrustify: %.c | _format_check_unstaged
        $(Q)$(UNCRUSTIFY) -f $< -o $@
-       $(Q)cmp -s $< $@ || (if [ "$$CI" == "true" ]; then diff -y $< $@; rm $@; exit 1; else cp $@ $<; fi)
+       $(Q)cmp -s $< $@ || (if [[ "$$CI" == "true" ]]; then diff -y $< $@; rm $@; exit 1; else cp $@ $<; fi)
 $(BUILDDIR)/format/%.uncrustify: %.cc | _format_check_unstaged
        $(Q)$(UNCRUSTIFY) -f $< -o $@
-       $(Q)cmp -s $< $@ || (if [ "$$CI" == "true" ]; then diff -y $< $@; rm $@; exit 1; else cp $@ $<; fi)
+       $(Q)cmp -s $< $@ || (if [[ "$$CI" == "true" ]]; then diff -y $< $@; rm $@; exit 1; else cp $@ $<; fi)
 $(BUILDDIR)/format/%.uncrustify: %.cpp | _format_check_unstaged
        $(Q)$(UNCRUSTIFY) -f $< -o $@
-       $(Q)cmp -s $< $@ || (if [ "$$CI" == "true" ]; then diff -y $< $@; rm $@; exit 1; else cp $@ $<; fi)
+       $(Q)cmp -s $< $@ || (if [[ "$$CI" == "true" ]]; then diff -y $< $@; rm $@; exit 1; else cp $@ $<; fi)
 $(BUILDDIR)/format/%.uncrustify: %.cxx | _format_check_unstaged
        $(Q)$(UNCRUSTIFY) -f $< -o $@
-       $(Q)cmp -s $< $@ || (if [ "$$CI" == "true" ]; then diff -y $< $@; rm $@; exit 1; else cp $@ $<; fi)
+       $(Q)cmp -s $< $@ || (if [[ "$$CI" == "true" ]]; then diff -y $< $@; rm $@; exit 1; else cp $@ $<; fi)
```

fix it for you? (change the `[` to `[[`)

bradjc, 2017-07-11:
I suspect it would fail in Travis if there were actually changes to be made (it runs fine when uncrustify makes no changes). But it works for me after: https://stackoverflow.com/a/3411105/5464

PR #493 (2017-07-11): tools: use = rather than == for string equality in sh
Fixes #491.
commit 5dd6c318 (2017-07-11): tools: use = rather than == for string equality in sh

Fixes #491.
closed 2017-07-11 by bradjc