tock/tock #499: RF233 drops 1 byte of payload on receive + ensuing rabbit hole of bugs

bbbert, 2017-07-13:
In the process of testing the RF233 implementation by just using the radio_rx and radio_tx userland examples and debugging the payload, I noticed that the radio_rx app was losing exactly one byte of the payload from the tx app. Reading the source, RF233 spec, and 802.15.4 spec to find the issue, however, reveals even more issues. Here are the full gory details that Paul and I have found.

**TL;DR** Lengths, offsets are mixed up, MAC header parsing + userspace interface are incomplete.

The entire confusion can be summarized in the following diagram:
```
+---------+-----+-----+-------------+-----+
| SPI com | PHR | MHR | MAC payload | MFR |
+---------+-----+-----+-------------+-----+
\______ Static buffer rx/txed to SPI _____/
                \__ PSDU / frame length __/
\___ 2 bytes ___/
```

- MHR is 9 to 9 + 6 + 6 bytes depending on whether the source/destination addresses are short/long. Actually, this assumes that the source PAN ID is omitted, but we'll come back to this later.
- MFR is the CRC/FCS and is 2 bytes long.
- SPI command is 1 byte for RF233. Note that this is not the SHR. (synchronization header)
- SPI buffer length = 2 + frame length. Again, this is not the PPDU (PHY protocol data unit) because the SPI command is not the SHR.
- PHR byte is always the frame length in RF233 frame reads/writes.
- Frame length = PSDU (PHY service data unit) = MHR length + MAC payload length + MFR length (2 bytes).
- SPI command and PHR determine the offset at which a PSDU must be stored, which I suppose might differ between different radios, but for now we can leave it as a constant.

The current implementation of the radio driver confuses `radio::Radio::header_size` (which I understand is meant to be the MHR?) with PHR + MHR. To further add to the confusion, `radio::HEADER_SIZE` is actually SPI com + PHR + MHR.

kernel/src/hil/radio.rs:27:
```
pub const HEADER_SIZE: u8 = 10;
pub const MAX_PACKET_SIZE: u8 = 128;
pub const MAX_BUF_SIZE: usize = 129; // +1 for opcode
pub const MIN_PACKET_SIZE: u8 = HEADER_SIZE + 2; // +2 for CRC
```
As we can see, `HEADER_SIZE` should really be `MIN_MHR_SIZE = 9`, and we should have something else like `MIN_PAYLOAD_OFFSET` be 11. `MAX_BUF_SIZE` is correct (max MTU is 127, +2 for SHR). If we understand "packet" to be "frame", then `MAX_PACKET_SIZE` should be the max MTU, which is 127. `MIN_PACKET_SIZE` should then be the minimum MHR size + MFR size, which should be 9 + 2 = 11. I propose the following constants:
```
pub const MIN_MHR_SIZE: u8 = 9;
pub const MAX_MTU: u8 = 127;
pub const MFR_SIZE: u8 = 2;
// The following two are just convenient for checking if a frame fits in a kernel buffer
pub const MIN_FRAME_SIZE: u8 = MIN_MHR_SIZE + MFR_SIZE;
pub const MAX_FRAME_SIZE: u8 = MFR_SIZE;

pub const PSDU_OFFSET: u8 = 2;
// Used for initializing a suitably-sized kernel buffer
pub const MAX_BUF_SIZE: usize = PSDU_OFFSET + MAX_MTU;
// Used for checking if a buffer slice (containing SPI com, PHR and MHR) is long enough to contain at least the minimum MHR).
pub const MIN_PAYLOAD_OFFSET: u8 = PSDU_OFFSET + MIN_MHR_SIZE;
```
Next, let's look at how the RF233 code uses the old constants.

1. rf233.rs:1111. `header_size` returns either 10 or 10 + 6 or 10 + 6, which is 1 + MHR.
2. rf233.rs:1104. `payload_offset` returns 1 + `header_size`, which is 2 + MHR.
3. rf233.rs:1124. `packet_header_size` takes a "packet" (really the whole SPI buffer) and checks if its length is at least `HEADER_SIZE`, which is only 1 + MHR. This is off-by-one and should really be `MIN_PAYLOAD_OFFSET`.
4. rf233.rs:1142-1273. All the `packet_get_xxx` methods make the same mistake.

Now we can begin tracing what happens in the radio driver. Here's the transmit path:

1. In userspace, `allow` and `subscribe`, then `command`. The length here presumably is the payload length.
2. capsules/src/radio.rs:118. Here it's fine to assume the source and length addresses are short because they are. Note that we call `radio.transmit` with `transmit_len = len (payload length) + radio.header_size (1 + MHR)` being 1 + MHR + MAC payload.
3. rf233.rs:1294. Now `payload` should really be `buffer` because it isn't the MAC payload. Right away, `len + 2` in the bounds check presumably accounts for MFR, but it is still off by one because it lacks the SPI command. Then we call `prepare_packet` with the same length.
4. rf233.rs:885. `len + 2 - 1` is now correct as the frame length. Note that `tx_len` is set to be `len`, which is still 1 + MHR + MAC payload. Now some SPI communication and state machine transitions occur before we actually perform a frame write.
5. rf233.rs:533. RF233 enters TX_STATUS_PRECHECK1 state, which calls `frame_write` with `buf_len` = `tx_len`.
6. rf233.rs:846. Since `buf_len` is 1 + MHR + MAC payload, `op_len = buf_len + 1` sets `op_len` to the correct 2 + MHR + MAC payload. The missing MFR is fine because automatic FCS generation is enabled on the RF233, which causes the 2 bytes to be appended to the transmission in hardware. The PHR byte is still correct because that is the length of the frame buffer write, not the actual PSDU length.

Here's the receive path:

1. In userspace, `allow` the buffer and `subscribe` to the callback, then yield. Presumably, we expect that only the MAC payload will be copied to the buffer.
2. rf233.rs:629. RF233 enters RX_START_READING state, in which it issues a frame read of length 0 (which still provides the SPI command byte and the PHR byte).
3. rf233.rs:635. RF233 enters RX_READING_FRAME_LEN_DONE state, with frame length (PHR byte) in `result` variable. `len = result - 2 (FCS) + 1 (PHR)` is thus length of PHR + MHR + MAC payload.
4. rf233.rs:642. Now we compare len with `MIN/MAX_PACKET_SIZE`, which already doesn't make sense. `MAX_PACKET_SIZE` is correct, but `MIN_PACKET_SIZE` includes the SPI command, which it shouldn't. Then we call `frame_read` with `buf_len` being actually 1 + PSDU length.
5. rf233.rs:858. Now we calculate `op_len = buf_len + 1` which again ends up being the correct SPI transfer size, 2 + PSDU length. Note that the RF233 frame read access process actually sends 5 + frame length bytes where the three extra bytes at the end are LQI (link quality indication), ED (energy detection) and RX_STATUS. Initiating an SPI read without these 3 extra bytes simply drops them.
6. rf233.rs:658. RF233 sends the frame back over SPI, and we go into the RX_READING_FRAME_DONE state. `len` is again PHR + MHR + MAC payload. Then we call `client.receive`, which goes to the radio driver.
7. capsules/src/radio.rs:193. The radio driver now calls `radio.payload_offset(false, false)`, which basically assumes the source and destination addresses are short. Even if that is correct, `buf[offset..len as usize]` ends up dropping one byte of the MAC payload because `len` doesn't include the SPI command.
8. We now notice that there is basically no way to obtain the received payload length in userspace, or the source address for that matter. We should really be calling the radio receive callback with the payload length in one of the parameters, and possibly the source address too.
9. Furthermore, notice that no CRC check is performed. RF233 checks the CRC automatically and sets a flag in RX_STATUS (and also some register), but we drop that byte and ignore it. This means that basically we are vulnerable to any corruption.
10. Also, since there is no MAC address filtering, we are essentially sniffing for any and all packets.

So basically a bunch of errors are made because of the confusing naming of lengths. However, that's not all. There are also more problems in the MAC header parsing facilities in rf233.rs.

1. Firstly, recall that in the receive path, the radio driver assumes that the source and destination addresses are short. Really, what it should be doing is parsing the MHR to obtain its length. Hence, we should have something like a `radio::packet_payload_offset`. But let's take a closer look at `packet_header_size`.
2. rf233.rs:1124. `packet_header_size` just ends up calling `packet_has_src/dst_long`, which check the corresponding bits in the frame control field in the MHR. Then, it delegates to `header_size`. However, this is actually insufficient. Apart from the fact that `header_size` currently also includes the extra length byte, it turns out that the minimum MHR size of 9 also assumes the source PAN ID is omitted. If it is not omitted, that is an additional 2 bytes. In reality (802.15.4 spec), only the frame control field is o [... cut ...] d 6LoWPAN and the potential future need for UDP/TCP/ICMP packet construction/parsing, we should pull all this device-independent functionality out into libraries. We also need a consistent and clean interface for using these libraries, and settle on a convention for how byte arrays are passed. Currently 6LoWPAN assumes they are mutable u8 slices and transmutes them into `repr(C, packed)` structs, but a better way might be to make structs that wrap u8 slices and expose getters/setters. Any ideas?

bbbert, 2017-07-13:
Apologies, `MAX_FRAME_SIZE` should be `MAX_MTU` and everything should be `usize`.
```rust
pub const MIN_MHR_SIZE: usize = 9;
pub const MFR_SIZE: usize = 2;
pub const MAX_MTU: usize = 127;
pub const MIN_FRAME_SIZE: usize = MIN_MHR_SIZE + MFR_SIZE;
pub const MAX_FRAME_SIZE: usize = MAX_MTU;

pub const PSDU_OFFSET: usize = 2;
pub const MAX_BUF_SIZE: usize = PSDU_OFFSET + MAX_MTU;
pub const MIN_PAYLOAD_OFFSET: usize = PSDU_OFFSET + MIN_MHR_SIZE;
```

phil-levis, 2017-07-16:
Great work! It sounds like the transmit path is working fine, and the bugs are primarily in the receive path?

A few comments:

HEADER_SIZE was intended to be both the MAC and PHY headers, not just the MAC header. Since the two can't be decoupled, and MAC-layer functionality (e.g., payload length) requires the PHY header, I put the two together. The variable describes the minimum size needed to store the header of a 15.4 packet. 

The userspace library is pretty thin, yes; my assumption was that one can just do things like get the length of the packet, address types, etc., purely with library calls that don't need to traverse the syscall boundary. The length is encoded in the PHY header, after all. The current library was implemented assuming that you just want to send/receive data, but it makes sense for the userland library to have access to the full packet headers for this kind of inspection.

closed 2017-07-26 by alevy