tock/tock #517: Static Analysis

ppannuto, 2017-07-20:
As suggested in #514, we should consider adding a static analysis pass at some point.

I'm a little wary of adding more dependencies to the install chain, so making it optional, or maybe something that travis runs, would probably be good.

niklasad1, 2017-07-21:
Yes. makes sense perhaps wait for rust in user-space would solve this.

However, [scan-build](http://clang-analyzer.llvm.org/scan-build.html) comes with Clang at least on Linux and I guess it is the same on OSX and pre-built binaries for OSX can be found [here](https://clang-analyzer.llvm.org/installation.html).
 
Also [ClangCheck](http://clang.llvm.org/docs/ClangCheck.htmll) exist I haven't looked into the difference between them but works fine on my arch linux box.

```bash
$ scan-build gcc test.c
scan-build: Using '/usr/bin/clang-4.0' for static analysis
test.c:7:3: warning: String copy function overflows destination buffer
  strcpy(new_str, str);
  ^~~~~~~~~~~~~~~~~~~~
1 warning generated.
scan-build: 1 bug found.
scan-build: Run 'scan-view /tmp/scan-build-2017-07-21-114824-8155-1' to examine bug reports.
```
``` bash
$ clang-check test.c -analyze
Running without flags.
/home/niklasad1/test.c:7:3: warning: String copy function overflows destination buffer
  strcpy(new_str, str);
  ^~~~~~~~~~~~~~~~~~~~
1 warning generated.
```

alevy, 2017-07-21:
Seems good. The clang tools are indeed good I think and I also agree it shouldn't part of the normal build chain, just because, yeah... More dependencies. Running it on Travis makes sense.

ppannuto, 2017-07-21:
I spent a few minutes looking into whether this would be easy to integrate, the first trick is that because we're cross compiling, we have to point to the actual compiler / toolchain to use, so you need to invoke `scan-build` like:

    $ ~/bin/checker-279/bin/scan-build --use-cc=/usr/local/bin/arm-none-eabi-gcc --analyzer-target=arm-none-eabi make

But that didn't really seem to work out of the box:

```
  CC        ../../libtock/isl29035.c
In file included from <built-in>:1:
In file included from ./../../support/warning_header.h:5:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/stdio.h:64:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/sys/cdefs.h:761:2: error: Unsupported architecture
#error Unsupported architecture
 ^
In file included from <built-in>:1:
In file included from ./../../support/warning_header.h:5:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/stdio.h:67:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/_types.h:27:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/sys/_types.h:33:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/machine/_types.h:34:2: error: architecture not supported
#error architecture not supported
 ^
In file included from <built-in>:1:
In file included from ./../../support/warning_header.h:5:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/stdio.h:67:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/_types.h:27:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/sys/_types.h:55:9: error: unknown type name '__int64_t'
typedef __int64_t       __darwin_blkcnt_t;      /* total blocks */
        ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/sys/_types.h:56:9: error: unknown type name '__int32_t'
typedef __int32_t       __darwin_blksize_t;     /* preferred block size */
        ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/sys/_types.h:57:9: error: unknown type name '__int32_t'
typedef __int32_t       __darwin_dev_t;         /* dev_t */
        ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/sys/_types.h:60:9: error: unknown type name '__uint32_t'
typedef __uint32_t      __darwin_gid_t;         /* [???] process and group IDs */
        ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/sys/_types.h:61:9: error: unknown type name '__uint32_t'
typedef __uint32_t      __darwin_id_t;          /* [XSI] pid_t, uid_t, or gid_t*/
        ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/sys/_types.h:62:9: error: unknown type name '__uint64_t'
typedef __uint64_t      __darwin_ino64_t;       /* [???] Used for 64 bit inodes */
        ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/sys/_types.h:68:9: error: unknown type name '__darwin_natural_t'
typedef __darwin_natural_t __darwin_mach_port_name_t; /* Used by mach */
        ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/sys/_types.h:70:9: error: unknown type name '__uint16_t'
typedef __uint16_t      __darwin_mode_t;        /* [???] Some file attributes */
        ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/sys/_types.h:71:9: error: unknown type name '__int64_t'
typedef __int64_t       __darwin_off_t;         /* [???] Used for file sizes */
        ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/sys/_types.h:72:9: error: unknown type name '__int32_t'
typedef __int32_t       __darwin_pid_t;         /* [???] process and group IDs */
        ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/sys/_types.h:73:9: error: unknown type name '__uint32_t'
typedef __uint32_t      __darwin_sigset_t;      /* [???] signal set */
        ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/sys/_types.h:74:9: error: unknown type name '__int32_t'
typedef __int32_t       __darwin_suseconds_t;   /* [???] microseconds */
        ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/sys/_types.h:75:9: error: unknown type name '__uint32_t'
typedef __uint32_t      __darwin_uid_t;         /* [???] user IDs */
        ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/sys/_types.h:76:9: error: unknown type name '__uint32_t'
[... cut ...]
scan-build: The analyzer encountered problems on some source files.
scan-build: Preprocessed versions of these sources were deposited in '/var/folders/_t/61hmf1293xq508hf1ksgyp900000gn/T/scan-build-2017-07-21-130315-36041-1/failures'.
scan-build: Please consider submitting a bug report using these files:
scan-build:   http://clang-analyzer.llvm.org/filing_bugs.html
```

I don't really want to sink more time into this now, but it looks like it's not quite as simple to integrate as other projects :/

JayKickliter, 2017-09-25:
@ppannuto I run `scan-build` on all my embedded projects at work. The only maintainable solution I've found is to have separate compiler flags for `clang` and `arm-none-eabi-gcc`. A lot of the flags are common between the two compilers, so there's usually only a handful of compiler-specific flags. The trickiest part is that you have to `-isystem` Newlib's headers so clang doesn't complain about things like format strings not matching their arguments' sizes.

JayKickliter, 2017-09-25:
I can drop some specific makefile/cmake configs if anyone is still interested in this, I just need to sanitize them first.

alevy, 2017-09-25:
@JayKickliter that would be great!

It would be cool to be able to compile C userland with clang as well, which would obviate needing to maintain additional flags. Though short of figuring out how to compile newlib with clang (last I checked it's not straight forward), that might be a pipe dream given the different relocation strategies...

JayKickliter, 2017-09-25:
I usually just link the application compiled with clang with newlib compiled with arm-gcc. But I've never done it with PIC. Also, I only use clang for static analysis, and ship binaries built with gcc since they are always smaller.

alevy, 2017-09-25:
> But I've never done it with PIC
Yeah, unfortunately GCC and Clang use competing relocations using the r9 register if you need RAM to be PIC addressed and not PC relative.

> I eventually gave up and used cmake with passed in toolchain files.
@ppannuto might be emboldened to hear that

ppannuto, 2017-09-26:
Suffice it to say that I certainly would not be the person standing in the way of Tock leaving make far far behind :)

As for static analysis, yeah, that'd be awesome!

bradjc, 2018-06-12:
Moved to wiki

closed 2018-06-12 by bradjc