tock/tock #528: Driver Numbers: Re-number to clarify stability

bradjc, 2017-07-22:
This is a note here so it gets added to the 1.0 milestone. The discussion has largely happened on the mailing list or on the calls. A future PR will include more details about what changes the new driver number strategy will include.

ppannuto, 2017-07-24:
Bringing the mailing list content/discussion over here:

@alevy:
> I've started organizing our system call interface into a new table towards stabilizing the system call numbers and each driver's interface for a 1.0 release. My thought is to group similar drivers together and leave padding in the system call namespace between each group to expand the group. The high level groups so far are:
>
> - Base: basic drivers that most hardware platforms probably have, and which lots of generic apps would probably use.
> - Cryptography: high-level crypto related operations. There are currently only three, but we might imagine having a cryptographic hash (some HW supports it), asymmetric encryption (there are some peripheral chips which provide it) etc.
> - Communication: These are basic HW communication busses: raw UART, SPI, I2C. This is probably where we would stick USB interface/endpoints, a CAN bus etc.
> - Sensors: High-level, abstract sensors (rather than peripheral specific interfaces). Ninedof is the only complete one (thanks Brad!), but we have more or less similar interfaces for light and temperature sensing that we can generalize, and probably can do the same for pressure. An open question is what to do about chips that provide multiple things (the SI7021 provides temperature and humidity). Another is whether the coulomb counters (e.g. MAX17205 and LTC294x) fit into this category or not.
> - Misc: drivers that are either platform specific, don't fit into a particular category or are in early stages. This category should be viewed as basically unstable and temporary.
> - IPC: everything >=255 is IPC related.
> I have a few goals I tried to balance in organizing things this way:
> 
> Getting a new chip and/or board up and running should be fairly straight forward following the system call numbers. For example, if you have a totally new chip, the timer, UART, and GPIOs are probably the first controllers you'll want to implement support for.
> Standardized drivers should be relatively generic so we can write cross-platform apps. Not every platform will have a temperature sensor, but if possible we should be able to write generic apps for all platforms that do, regardless of the specific choice of peripheral chip.
> There should be room for expansion in each category. This influenced the ordering of the categories as my hunch is that sensor types are going to have the large space to expand, while we can probably enumerate today what might go in each other category.
... snip original layout ...
> What are folks thoughts on:
> 
> 1. This method of organizing system calls
> 2. The choice to use relatively general driver interface for standardized driver numbers
> 3. What should count as a sensor? (specifically are coulomb counters sensors or are they more likely to be used as a lower-level part of a system service?)

@ppannuto:
> > 1. This method of organizing system calls
> 
> I like it, I think it helps to make the system accessible. I'd rather have fewer categories and things that don't fit perfectly than the gamut of categories, which I think you've done well here.
> 
> > 2. The choice to use relatively general driver interface for standardized driver numbers
>
> I agree with this, but I think it means that you're missing a critical category, specifically the notion of "chip-specific" drivers. i.e. I would imagine that my accelerometer driver is accessible both via the "standard" 9-dof interface, and (an optional) chip-specific driver number for its bonus features.
> 
> I feel like we should have a sparser number space than the current allocations. Is there a reason/need to compact everything non-IPC to < 255? I would lean towards making IPC >= 65536 or something like that.
>
> > 3. What should count as a sensor? (specifically are coulomb counters sensors or are they more likely to be used as a lower-level part of a system service?)
>
> I'd call them a sensor

@phil-levis:
> Standardizing/cleaning this up is a great idea. But I’d suggest segmenting on the high bits, rather than small integers.
>
> IIRC, our original formulation for device numbers was that devices with a 0 in the MSB were well-defined, standard, interfaces, while those with a 1 in the MSB were board/platform specific, free-for-all. I.e., we manage the namespace with 0 in the MSB in the main repository.
>
> I’d suggest allocating more than 255 for not IPC. E.g., following the above address convention:
>
> 00x: standard, non-IPC
> 01x: standard, IPC
> 10x: custom, non-IPC
> 11x: custom, IPC
>
> Basically, I’d suggest that given we have 32 bits, split up the namespace in the more significant bits so we’re sure we have lots of space.
>
> Sensors will almost certainly be the part of the namespace that has the most entries.

@alevy:
> > But I’d suggest segmenting on the high bits, rather than small integers.
> Agreed.
> …
>
> > IIRC, our original formulation for device numbers was that devices with a 0 in the MSB were well-defined, standard, interfaces, while those with a 1 in the MSB were board/platform specific, free-for-all. I.e., we manage the namespace with 0 in the MSB in the main repository.
> Yeah, you're right.
> …
> > I’d suggest allocating more than 255 for not IPC.
> Slight clarification, all of IPC actually only takes up a single driver number---each service is named by a string (e.g. "org.tockos.rot13"), which gets assigned a PID at runtime.
> 
[... cut ...]
> Yes this makes sense. Assuming we don't actually need more than one for IPC, how about we just stick it as one of the driver numbers in the "Common" category, and extend your notation as follows:
> 
> 1. The 1st MSB signifies 0 for standard, 1 for custom.
> 2. If standard, the next 15 MSBs signify the category. This gives us up to 32k categories, which should be plenty (famous last words)
> 3. The bottom 16 bits are for each specific driver in the category, giving us up to 64k drivers per category, which should also be plenty (the previous were my famous last words, so these will go unnoticed).

ppannuto, 2017-07-24:
Inspired by the title of this issue, should we also have a mechanism for marking drivers as "stable" or "unstable" (i.e. ABI locked in / permitted to change)? I could imagine the second-most significant bit for this, as in:

```
00x: standard, stable
01x: standard, unstable
10x: custom, stable
11x: custom, unstable
```

ppannuto, 2017-07-24:
On a related thought (that may be introducing too much complexity): should drivers be versioned?

Currently we define call `0` for everything to indicate driver availability/presence. Maybe instead of just returning success, it should be `SuccessWithValue` where the value is the version number of the driver (and thus supported features of the driver). In my imagining of this, this allows us to add (but not remove) additional features to stable drivers

ppannuto, 2017-07-24:
> Why have "standard but unstable" instead of just "custom that becomes standard"

My thought here was that "standard but unstable" is a mechanism to express that this is an "official" driver supported by Tock core developers, but it's under development still, as opposed to "custom" drivers which are wholly unowned / unaffiliated with the core team. Perhaps that's an unnecessary distinction

bradjc, 2017-07-25:
Proposal:

```
High level:

   ┌─────────── [ 4 bits] Stability and custom/standard.
   │  ┌──────── [12 bits] Driver category.
   │  │   ┌──── [16 bits] Individual driver.
   │ ┌┴┐ ┌┴─┐
0x 0 000 0000

Driver number by bit:

   3                   2                   1                   0
 1 0 9 8 7 6 5 4 3 2 1 0 9 8 7 6 5 4 3 2 1 0 9 8 7 6 5 4 3 2 1 0
+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
|C|S|Res| Category              | Driver                        |
+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+

- C: Core (in tock tree) = 0, custom = 1.
- S: Stable = 0, unstable = 1.
- Res: Reserved. Must be 0.
- Category: The broad category of the driver.
- Driver: Which actual driver it is.

Categories
----------

0x000: Base:         Core embedded things like LEDs, GPIOs, timer, ADC, button, DAC, console.
0x001: Kernel:       Kernel things, like IPC.
0x002: HW Buses:     Raw access to I2C, SPI, UART, etc.
0x003: Radio:        Wireless radios.
0x004: Cryptography: AES, CRC (?), RNG, etc.
0x005: Storage:      Flash, FRAM.
0x006: Sensors:      Generic sensor drivers (e.g. light sensor).
0x007: Sensors2:     Specific sensor drivers (e.g. SomeCrazyIC7561).
0x008: Other ICs:    Coulomb counters, I2C extenders.
```

### Most significant four bits

Highest bit is as described before, and separates drivers in the tock repo from those that are project specific.

Next bit separates stable drivers (or partially stable drivers) from unstable ones. I think it makes more sense to not just put our unstable drivers in the custom (0x8XXXXXXX) space.

Four bits rather than 2 so that it is clearly readable (one character in hex), and we are unlikely to need more than 4k categories...

ppannuto, 2017-07-25:
I like this proposal (with the sole caveat of bikeshedding `Sensors2` -> `Sensor ICs`)

phil-levis, 2017-07-25:
> On Jul 24, 2017, at 7:28 AM, Pat Pannuto <notifications@github.com> wrote:
> 
> Inspired by the title of this issue, should we also have a mechanism for marking drivers as "stable" or "unstable" (i.e. ABI locked in / permitted to change)? I could imagine the second-most significant bit for this, as in:
> 
> 00x: standard, stable
> 01x: standard, unstable
> 10x: custom, stable
> 11x: custom, unstable

Hm. I think stability/instability shouldn’t affect the device number you use (and therefore break applications). If you want to distinguish them, I’d instead suggest doing so by their location in the tree. It’s something you care about when compiling the kernel, not when you’re writing your application (where it will be hidden anyways under sys calls).

Phil

bradjc, 2017-07-25:
> I think stability/instability shouldn’t affect the device number you use (and therefore break applications).

Are you thinking that while a driver is unstable its interface _could_ change, but likely won't change, and therefore we should opportunistically have applications keep working in this common case, rather than ensuring they won't work by changing the driver number?

I've been thinking for a while that we need a way to note what is old, established, stable code from what is newer experimental code. Maybe continuing to expand the `//!` header to include a "stability" section would be a good solution (inspired by, for example, https://nodejs.org/api/timers.html).

phil-levis, 2017-07-26:
I'm thinking of the case when I have a driver whose interface is stable but it's still marked as unstable due to implementation issues. Should I have to recompile my app to use the new driver number when the implementation issues are fixed?

phil-levis, 2017-07-26:
But it seems like our imagined use cases are different -- are these unstable numbers just basically a way to have a driver number that doesn't yet occupy the normal namespace? For example, if we said that master had no such drivers in it, then that would be another way. I think managing stable/unstable with branches is probably easier than having unstable things hanging around in master. To some degree, it's not really in stone until there's a TRD on it.

ppannuto, 2017-07-26:
> I'm thinking of the case when I have a driver whose interface is stable but it's still marked as unstable due to implementation issues.

Ah, I had a different notion of stability in mind. Specifically I was imagining this bit to mark ABI stability. An app using a stable driver should never need recompilation if the driver changes, that's the contract

> Should I have to recompile my app to use the new driver number when the implementation issues are fixed?

One could imagine a code path that if driver lookup fails then checks for `driver_num & 0xbfffffff`, emitting a `debug!{"Warn driver {} has been stabilized, please double-check the interface and update this app to use the stable driver number"}`, and trying to call the new stable driver

> But it seems like our imagined use cases are different...

I think the things that are important to me:

  - It's clear to users what drivers are stable, reliable, (expected to be) working, supported
  - It's clear to developers what the path to adding a stable driver is
  - Stable drivers should remain backward compatible (an app using a stable driver should never need recompilation)
  - There's a low-barrier to introducing an initial/"unstable" driver 
  - There's a mechanism to identify stability from crash dumps
    - i.e. Linux's kernel dump will indicate if there are any tainted modules loaded, and anything in `drivers/staging/` is automatically tainted

ppannuto, 2017-07-26:
> Maybe continuing to expand the `//!` header to include a "stability" section

What I don't love about putting it in a comment is there's no way to get it back from a compiled image (i.e. a crash dump), I could certainly imagine a future dump to the effect of

```
Tock Kernel Dump
...
loaded capsules:
   - spi - stable
   - mywhizdig - unstable!
...
```

Though maybe the solution there is simply to add a `stability` field to the Driver trait instead/in-addition-to the comment

bradjc, 2017-07-27:
Update:

```
High level:

   ┌─────────── [ 4 bits] Custom/standard.
   │  ┌──────── [12 bits] Driver category.
   │  │   ┌──── [16 bits] Individual driver.
   │ ┌┴┐ ┌┴─┐
0x 0 000 0000

Driver number by bit:

   3                   2                   1                   0
 1 0 9 8 7 6 5 4 3 2 1 0 9 8 7 6 5 4 3 2 1 0 9 8 7 6 5 4 3 2 1 0
+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
|C|Reser| Category              | Driver                        |
+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+

- C: Core (in tock tree) = 0, custom = 1.
- Res: Reserved. Must be 0.
- Category: The broad category of the driver.
- Driver: Which actual driver it is.

Categories
----------

0x000: Base:         Core embedded things like LEDs, GPIOs, timer, ADC, button, DAC, console.
0x001: Kernel:       Kernel things, like IPC.
0x002: HW Buses:     Raw access to I2C, SPI, UART, etc.
0x003: Radio:        Wireless radios.
0x004: Cryptography: AES, CRC (?), RNG, etc.
0x005: Storage:      Flash, FRAM.
0x006: Sensors:      Generic sensor drivers (e.g. light sensor).
0x007: Sensor ICs:   Specific sensor drivers (e.g. SomeCrazyIC7561).
0x008: Other ICs:    Coulomb counters, I2C extenders.
```

Basically removed the stable/unstable bit so that we don't force applications to break even if the API doesn't change.

I do like the idea of adding a `type` field (or something like it) to the `Driver` type with an enum for stability.

alevy, 2017-08-03:
I think we're in more or less agreement, so let's move the discussion to PR #559

closed 2017-08-03 by alevy