tock/tock #529: nRF52-DK hangs while loading the user-app

niklasad1, 2017-07-23:
Hi,
After [#511](https://github.com/helena-project/tock/pull/511)
nRF52-DK does not behave as expected in the master branch (note it does not support BLE yet)

For example when I flash [printf_long](https://github.com/helena-project/tock/tree/master/userland/examples/tests/printf_long) to board it hangs and never gets any interrupts etc:
```
TOCK_DEBUG(0): src/main.rs:301: Initialization complete. Enterin

```
But when I flash [ble_scan](https://github.com/helena-project/tock/tree/master/userland/examples/ble_scan) it works as expected:
```
TOCK_DEBUG(0): src/main.rs:301: Initialization complete. Entering main loop
BLE Scanner

ble_adv_start error -11
```
Without an user-app loaded it works fine as well
```
TOCK_DEBUG(0): src/main.rs:301: Initialization complete. Entering main loop
```
Can anyone  reproduce this behaviour?

alevy, 2017-07-23:
I'll take a look tomorrow or Tuesday when I'm near hardware again.

niklasad1, 2017-07-23:
Sounds good, nRF51 seem to behave unexpected as well.

closed 2017-07-27 by niklasad1