tock/tock #530: Allocate the PROCS structs in a special grant region of each process.

bradjc, 2017-07-24:
This would eliminate the fixed size list of processes and allow an arbitrary number of processes.

ppannuto, 2017-07-24:
Would this mean that the process memory allocation would have to be
round_up_to_nearest_power_of_2ish(APP_HEAP + KERNEL_HEAP + sizeof(Proc))

bradjc, 2017-07-25:
`round_up_power2(app.header.minimum_ram_size + sizeof(Proc))`, but yes.

alevy, 2017-07-25:
Would PROCS then be `[Option<`static mut Process>; NUM_PROCS]`?

cbiffle, 2017-07-26:
A linked list, of course, would cause every conversion of an `app_id` to a `&Process` to run in O(n) time (linear in the number of processes). Depending on your real-time constraints this might be a bad idea. You could get it down to O(n log n) using a skiplist with slightly more memory overhead.

alevy, 2017-07-26:
I'm with @cbiffle on this in terms of iteration complexity, as well as have some other doubts about making a linked list work, but I do think it's worth trying out.

bradjc, 2018-05-03:
Just merged this.

closed 2018-05-03 by bradjc