tock/tock #533: program and flash require "--arch cortex-m4" for my board

daniel-scs, 2017-07-25:
I was able to program an app by hacking the flag into `userland/Program.mk`, and to flash the kernel by doing the same in `boards/hail/Makefile`, but I'm not sure quite how to solve the problem in general.  Perhaps the arch can be assumed for Hail, if the bootloader attributes don't specify it?

(My board says `12-2016 [A]`.)

(I love tockloader.)

brghena, 2017-07-25:
I'm confused what the problem you're having is.

The `boards/hail/Makefile` already has the proper flags (including `--arch cortex-m4`) https://github.com/helena-project/tock/blob/master/boards/hail/Makefile#L18

As for the userland stuff, tockloader should be reading them off of the Hail and using them.

It seems likely that your bootloader is just really old. Have you tried connecting JTAG to the SAM4L and running `make flash-bootloader` to update it?

daniel-scs, 2017-07-25:
Yeah that's what I figured ... but no tengo tiny tiny JTAG pin-contactor-thing.

(If you know offhand what the connector is called, I'll order us one here.)

bradjc, 2017-07-25:
http://www.tag-connect.com/TC2030-IDC-NL

Do you have an adapter board (from massive 20 pin ARM JTAG to the 6 pin 0.1" tag connect header)?

daniel-scs, 2017-07-25:
Yep, we have the 20-pin-to-6-pin thing, which we use for the legged JTAG connection on the imix.

So this legless thing works, huh?  Sounds good, thanks.

bradjc, 2017-07-25:
If you are willing to convert a legged adapter to a legless one you can easily snap off the legs. I always buy the legged ones, and then make my own legless ones on demand.

daniel-scs, 2017-07-25:
Ha, yes I finally realized that one connector is a subset of the other.  And I'm glad that I'll be able to use the quick-press flashing method on imix boards as well, because getting the legs in and out is awkward -- and now we've got tockloader for everyday programming.

Anyway I had already ordered a legless one, so no violence will be necessary for now ...

ppannuto, 2017-07-26:
I just updated your message to indicate that updating the bootloader can alleviate the need to pass flags ( https://github.com/helena-project/tockloader/commit/712a1991cc5d63c1f2e4ebdd8789767dfbd76e73 )

I think that should close this issue

closed 2017-07-26 by ppannuto