tock/tock #536: 11th queued callback will be dropped without warning or error

cbiffle, 2017-07-26:
Just started reading your code, so my apologies in advance if I missed something fundamental.

The `schedule` operation on `Callback` returns a `bool` to indicate whether the callback was successfully scheduled. Based on my initial review of capsule code, nobody seems to check it.

For example, [SPI does not check it](https://github.com/helena-project/tock/blob/master/capsules/src/spi.rs#L285), nor does [IPC](https://github.com/helena-project/tock/blob/master/kernel/src/ipc.rs#L56).

Switching from `bool` to a type bearing `#[must_use]` (such as `Result`) is probably the quickest fix, but it's not immediately clear to me what can be done at the point where such an error would be detected. (Other than `panic`.) The app is probably already waiting for the event, and cannot reasonably respond.

It might be better to keep track of the number of outstanding `Callback` values per app. If an app creates more than the limit (currently 10), the `schedule` syscall could report an error to the *app*, which is probably the right place to respond to it. (`Callback` would then decrement the associated counter using a `Drop` impl.)

Obviously kernel-internal callbacks would not be subject to this limit.

Thoughts?

Thanks!

alevy, 2017-07-26:
The (not documented anywhere) thought we had originally was that processes would be able to choose how large their `Callback` buffer is, and so it would be up to a processes to size it correctly. That was never implemented, but it should actually be fairly straight forward to add to the binary format header now (right @bradjc?), and pretty trivial to implement (since the ring buffer is allocated in the process's grant region). That doesn't really address the concern you raised, but it's a bit of background.

> It might be better to keep track of the number of outstanding Callback values per app. If an app creates more than the limit (currently 10), the schedule syscall could report an error to the app, [...]

Intuitively, this seems right to me, and would be pretty easy to implement (just have `schedule` check `is_full`). Though it would be good to flesh out what an app might actually do with that notification...

alevy, 2017-07-26:
Adding this to the Tock 1.0 milestone because I'm realizing this almost certainly impacts the system call interface (i.e. by adding an extra callback or something) and it might be an important thing to address soon.