tock/tock #538: Drop semantics for Owned<T> seem questionable

cbiffle, 2017-07-26:
`Owned<T>` is produced by operations like `enter` on a `Container`. I notice that it has a `Drop` impl, which (for app containers) winds up calling `app.free(...)` [here](https://github.com/helena-project/tock/blob/master/kernel/src/container.rs#L69).

It so happens that `free` on a `Process` is [currently a no-op](https://github.com/helena-project/tock/blob/master/kernel/src/process.rs#L1127) without documentation, but I assume based on the name that it's intended to free the associated buffer.

But a new `Owned<T>` is created and dropped for each container access. For example, during `enter`, it gets dropped [here](https://github.com/helena-project/tock/blob/master/kernel/src/container.rs#L190). This means you'll call `free` on every container access.

Is this intentional?

alevy, 2017-07-26:
Thanks @cbiffle, I think your general assessment is 100% correct (`free` is currently unimplemented pending a decent allocation implementation), and we don't want drop to be called when the `Owned` for a root pointer goes out of scope---because conceptually it's not really going out of scope, that's just an artifact of how we implement `enter`.

We do, however, want `Drop` for `Owned` values created using `Allocator#alloc`. I think the missing piece is basically a `mem::forget(root)` [here](https://github.com/helena-project/tock/blob/master/kernel/src/container.rs#L178), [here](https://github.com/helena-project/tock/blob/master/kernel/src/container.rs#L190) and [here](https://github.com/helena-project/tock/blob/master/kernel/src/container.rs#L209).

I could have missed a spot, but the idea is that we're regenerating a root `Owned` from a raw pointer every time we enter a container. That shouldn't be dropped. However, a container may also have other `Owned` values (e.g. in a field inside of it's root struct) that can be created using the `Allocator`. If those go out of scope (e.g. if the field is an `Option<Owned<T>>` and is replaced with a `None`), we _should_ call drop.

Does that seem reasonable?

cbiffle, 2017-07-26:
Hm, covering every code path with `mem::forget` seems error-prone to me. If you define `Owned<T>` as having semantics similar to a kernel-segment-allocated `Box<T>` (i.e. its backing memory will be freed on drop), then you probably don't want to use it at all for `enter` and `each`. Really, you're handing the closure a borrowed exclusive reference.

I'd probably change the signature of `enter` to take

```rust
F: FnOnce(&mut T, &mut Allocator) -> R
```

instead.

alevy, 2017-07-26:
> Hm, covering every code path with mem::forget seems error-prone to me

Agreed.

> I'd probably change the signature of enter to take [...]

I think you're right.

Can anyone else think of why we used `Owned` for the `enter` closure in the first place? It could have been just a mistake. I can't justify it now.

ppannuto, 2017-07-26:
I don't think we ever thought through it that much, from https://github.com/helena-project/tock/commit/c05368ed3574b22120ffcf44271f32ab1c1faa32, it looks like that was part of the OSDI sprint last year

I recall thinking the interface was awkward when I was working in the `debug!` stuff a while ago too (though that was more around requiring an `Allocator`), but perhaps it's worth revisiting the Container interfaces more deeply at some point.

I agree that dropping `Owned` on the usages is a good step

PR #542 (2017-07-26): container: stop using Owned with enter, each
This introduces another smart-pointer type, Borrowed, for the case where the
memory is not actually owned but borrowed. I would have preferred to just use
`&mut T`, but capsules are relying on the value including an `app_id`, so this
seemed like the smallest delta.

Fixes #538.
commit 051f627b (2017-07-26): container: stop using Owned with enter, each

This introduces another smart-pointer type, Borrowed, for the case where the
memory is not actually owned but borrowed. I would have preferred to just use
`&mut T`, but capsules are relying on the value including an `app_id`, so this
seemed like the smallest delta.

Fixes #538.
closed 2017-07-26 by alevy