tock/tock #544: tockloader to flash kernel for nRF51 & nRF52

niklasad1, 2017-07-27:
Hi,
I think we should migrate to use ```tockloader``` to flash the kernel for nRF51 & nRF52.

alevy, 2017-07-27:
I agree. The current bootloader for the SAM4L is written using the Atmel SDK, so it wouldn't easily translate to the NRF5x. I believe the only necessary pieces to write it natively in Tock are the UART, CRC and the non-volatile memory controller. The NRF5x series doesn't seem to have a CRC controller, but we can just implement that in software (it's not ideal, but it's not so bad either).

We have all of those pieces already for the SAM4L now, so we would be able to implement a cross platform bootloader.

bradjc, 2017-07-27:
In the current bootloader the CRC is done in software, just FYI

https://github.com/helena-project/tock-bootloader/blob/master/src/crc32.c

bradjc, 2018-11-29:
Support for an nrf52 bootloader was just merged! It doesn't really make any sense to use it on the nrf52dk because:

1. Something on my computer corrupts messages longer than 64 bytes, which may have something to do with the serial port on mac being a "usbmodem" port and this issue: https://devzone.nordicsemi.com/f/nordic-q-a/16721/pca10040-uart-problem
2. `make flash` works over USB with no additional hardware.

So, when we have a board (likely with an ftdi chip) that needs a bootloader we can build from what we have (shouldn't take much).

closed 2018-11-29 by bradjc