tock/tock #561: SPI chipselect

drahnr, 2017-08-03:
In the code for chip select for SPI in capsules, I see

```
            3 /* set chip select */ => {
                // XXX: TODO: do nothing, for now, until we fix interface
                // so virtual instances can use multiple chip selects
                ReturnCode::ENOSUPPORT
            }
            4 /* get chip select */ => {
                // XXX: We don't really know what chip select is being used
                // since we can't set it. Return error until set chip select
                // works.
                ReturnCode::ENOSUPPORT
            }
```

what is necessary to implement this properly? Is there some additional information available?

alevy, 2017-08-03:
This interface is pretty unsable still. What you're pointing to is the system call driver for access a SPI device directly from userspace. Kernel components, like this one, can either access something that implements the `kernel::hil::spi::SpiMaster` trait, for an entire SPI bus + chip selects, or that implements the `kernel::hil::spi::SpiMasterDevice` trait, for a specific peripheral that is selected by a particular chip select.

The current SPI system call driver uses a `SpiMasterDevice`, which means it fundamentally only has access to a multiplexed SPI bus mediated by particular chip selects---chip select managed by a virtualization layer hidden from the driver at this level.

What that means is that a userspace process interacting with this driver is basically able to fire of SPI transactions to a _particular_ peripheral device connected to the SPI bus (without needing to coordinate with other components using the same bus), but cannot actually decide which chip select to use.

Why are these interface stubs there? Good question... it's because they used to be useful a while ago when the driver was structured differently and when we actually used SPI directly from userspace (it was for a port of an rf233 driver, which is now in the kernel).

So implementing these commands would mean changing the `Spi` driver to use a `SpiMaster` instead of `SpiMasterDevice` and probably changing some implementation details.

It's certainly possible to implement, but I'm not sure, at the moment, if that's something we want or not. It depends on what are reasonable use cases for userspace processes to access the spi bus directly, rather than through a virtualized, higher level interface such as a peripheral specific driver.

The main question this brings up is whether the interface for userspace should look like it does in Linux, where a particular SPI device is surfaced as a virtualized device (e.g. `/dev/spi0.1` is the SPI peripheral on bus 0 with chip select 1), or instead have a single interface for a whole SPI bus and the user performs chip select themselves.

I _think_ the former (what Linux does) is simpler to share across processes, abstract over peripherals connected to different SPI buses, as well as to program against, but I don't have specific use cases to motivate that. Thoughts @drahnr?

This is related, btw, to #559 where we are starting to actually stabilize and agree on certain common system call interfaces. SPI is one of them.

drahnr, 2017-08-04:
Thanks for the wealth of information!

I am all in favor for the `spi${bus}.${cs}`, keeps simple and knowing linux little new things have to be learned and eases adoption.

What I am looking for is `kernel::hil::spi::SpiMasterDevice` and then implementing drivers for my specific devices on top of that.

Given that, I assume I have to expand the ABI, correct? How would I do this or is there a way around modifying the ABI yet adding custom functionality to the kernel?

alevy, 2017-08-04:
> Given that, I assume I have to expand the ABI, correct? How would I do this or is there a way around modifying the ABI yet adding custom functionality to the kernel?

It what you want to do with it. For example, if you're SPI peripheral fits into one of the existing system call drivers or if the functionality is otherwise hidden from userspace, it's not necessary to touch the ABI. A decent example of this right now is `usbc_client` in the `capsules` crate or the `i2c_dummy` kernel "app" in `boards/imix/src/i2c_dummy` (that one's just a test, but it's an app that doesn't interact with userspace at all).

_However_, extending the system call interface is _not bad_. Take a peek [here](https://github.com/helena-project/tock/issues/528#issuecomment-317602068) for what the system call driver number is going to look like (in the next couple weeks), but basically if you set the high bit it counts as a "custom" (i.e. considered board specific) driver and you can do whatever you want.

To actually _implement_ it, you'll want to implement the `kernel::Driver` trait (e.g. the `crc` capsule is a reasonably comprehensive example) and then map to it from in the platform's `with_driver` method, as [here](https://github.com/helena-project/tock/blob/master/boards/hail/src/main.rs#L75).

If this doesn't answer the question, happy to answer more. If it does and you have more "how to implement" questions, can we close this issue and move it to the [mailing list](https://groups.google.com/forum/#!forum/tock-dev)? Thanks!

drahnr, 2017-08-05:
Thank you so much! I will dig in and open upcoming questions on the mailing list. Sorry for derailing this ticket a little..

closed 2017-08-05 by drahnr