tock/tock #564: [nrf52dk] Led Demos do not work

drahnr, 2017-08-08:
```
make TOCK_BOARD=nrf52dk 
make -C boards/nrf52dk/
make[1]: Entering directory '/home/bernhard/Projects/tock/boards/nrf52dk'
   Compiling nrf52dk v0.1.0 (file:///home/bernhard/Projects/tock/boards/nrf52dk)
   Compiling kernel v0.1.0 (file:///home/bernhard/Projects/tock/kernel)
   Compiling gcc v0.3.51
   Compiling cortexm4 v0.1.0 (file:///home/bernhard/Projects/tock/arch/cortex-m4)
   Compiling capsules v0.1.0 (file:///home/bernhard/Projects/tock/capsules)
   Compiling nrf52 v0.1.0 (file:///home/bernhard/Projects/tock/chips/nrf52)
    Finished release [optimized + debuginfo] target(s) in 7.91 secs
   text	   data	    bss	    dec	    hex	filename
  86512	   1756	  13160	 101428	  18c34	target/thumbv7em-none-eabi/release/nrf52dk
make[1]: Leaving directory '/home/bernhard/Projects/tock/boards/nrf52dk'
```

uninstalling just in case there are some remainders from before, just to confirm.
```
tockloader uninstall --jtag --board nrf52dk --arch cortex-m4    
Removing app(s)  from board...
No apps are installed on the board
```

installing the blink app (or the buttons app, same for both)
```
tockloader install --jtag --board nrf52dk --arch cortex-m4 boards/nrf52dk/apps/blink/build/blink.tab 
Installing apps on the board...
Finished in 1.399 seconds
```

But the list does not yield this app being installed....
```
tockloader list --jtag --board nrf52dk --arch cortex-m4
No found apps.
Finished in 0.329 seconds
```

```
tockloader --version
0.7.2
```
I used today's git `5aea4788` connected and powered via USB. None of the Leds change their colour. The debugger does not show any crashes, which makes sense without any apps.

This does not seem right to me, or at least I can not see what I am doing wrong.

niklasad1, 2017-08-08:
@drahnr 

1. You need to flash the kernel to the board by ```make TOCK_BOARD=nrf52dk flash```
2. ```$ cd userland/examples/blink```
3. ```make TOCK_BOARD=nrf52dk flash```

FYI, I don't know why tockloader option don't work. I haven't got to work on nrf51dk either whereas it works for the other guys see [#15](https://github.com/helena-project/tockloader/issues/15).
I thought it was just my installation which was broken.

Let me know whether it works or not.

drahnr, 2017-08-08:
I think I mixed it up with this line from the getting started `Alternatively, apps can be built and automatically uploaded from the Tock root directory:` followed by `make path/to/app`.

I will verify it tomorrow.

niklasad1, 2017-08-08:
Yes, you can flash userapps from the root directory using:
```$ make examples/blink```

But the ["root-makefile"](https://github.com/helena-project/tock/blob/master/Makefile#L51-#L53) only support the ```program option``` which is using tockloader under the hood and it is not currently for supported for nRF5X.

Ping @ppannuto should we support ```flash option``` ["root-makefile"](https://github.com/helena-project/tock/blob/master/Makefile) 
(Otherwise we have add support for nRF52DK in boards to enable the ```program option ```)

Thus, I suggest to test according the sequence as I stated earlier 😃

drahnr, 2017-08-09:
@niklasad1 that works.

My mistake was going into

`Projects/tock/boards/nrf52dk/apps/blink`                                                                                                                                                   

```
exa -la 
lrwxrwxrwx@ 32 bernhard  2 Aug 12:12 blink -> ../../../userland/examples/blink
lrwxrwxrwx@ 34 bernhard  2 Aug 12:12 buttons -> ../../../userland/examples/buttons
lrwxrwxrwx@ 34 bernhard  2 Aug 12:12 c_hello -> ../../../userland/examples/c_hello
```
were linked and apparently and tried to use the `tockloader` directly.

Thanks a lot!

niklasad1, 2017-08-09:
All right, good

I will close this issue but open a new issue related to your questions/issues.

Sorry, it is not good that the program option don't work at the moment for nRF5X boards and don't comply the getting started guide.

closed 2017-08-09 by niklasad1