tock/tock #576: Remove imixv1?

alevy, 2017-08-10:
@shaneleonard kindly handed out new imixes to everyone who was at the Intel event this week. Of the people that have imixv1s, I think that only excludes @brghena and @bradjc. Can we remove `imixv1` from the repo?

shaneleonard, 2017-08-10:
I haven't yet given a board to Phil, but will do that by the end of the day today. Also I believe Dom has an imixv1, but I don't know if he's actively using it. The only other people I can think of are Maurice and Petar, but I'm assuming they aren't developing anything anymore?

I gave @ppannuto and @adkinsjd 4 boards total, enough for @brghena and @bradjc, and it sounds like they might want one more for Neil, but he doesn't currently have a v1.

shaneleonard, 2017-08-10:
Also I gave one to Holly at Stanford during the school year, she was working with @daniel-scs, so if she picks it back up in the fall I can give her a new board.

phil-levis, 2017-08-10:
I think we can safely remove imixv1 from the repository -- continuing to maintain it will be more work than giving people v2 boards :)

PR #577 (2017-08-10): Remove imixv1
Remove the legacy imixv1 board from the repository. New and improved imix v2.0s are out and distributed and we don't want to support imix v1.* anymore.

Fixes #576
closed 2017-08-10 by alevy