tock/tock #590: Cleaner register handling interface

shaneleonard, 2017-08-16:
As I'm writing the Teensy port, it would be really nice to have a register/bitfield interface that's better than just VolatileCell and manual bit-twiddling. @daniel-scs implemented a pretty nice interface for his USBC implementation (in sam4l/usbc/common/), and it would be helpful for me (and hopefully everyone) if we decided whether to adopt it as the global standard (i.e. move it to kernel::common) and whether to change it.

My wishlist:

1. Ergonomic to define registers and read/modify/write their individual fields
2. Sensible naming/namespacing and self-documenting; it's easy to read the code and understand what values are being set in which fields in which registers
3. Support for variable-width registers (u8, u32)
4. Differentiation between read-only, write-only, and read-write
5. Minimal code bloat, and as efficiently compiled as possible (eg close to the efficiency of defining constants and manually bit-twiddling)

@daniel-scs's implementation already has these qualities, although I have a couple of complaints (which may or may not be a big deal):

1. Bitfields are tied to a specific memory address. It's often useful to be able to read/write a bitfield at arbitrary locations.
2. There's no way to write multiple fields with a single volatile store. This can sometimes be necessary, but I don't know if it's necessary on our current platforms.
3. Registers aren't arranged into a C-like struct. The ability to define a struct of registers at an arbitrary base address is very useful, especially for peripheral blocks that have duplication
4. Lots of different traits and types--not necessarily bad, I just don't know what the code compiles down to

shaneleonard, 2017-08-17:
EDIT: @daniel-scs's version doesn't actually support byte-width registers

brghena, 2017-08-28:
This is a really good idea, although I don't have any clear ideas on how it would be implemented.

I think your note about writing multiple fields with a single store is definitely important, both for efficiency reasons but also for correctness with some hardware.

Some things you might want to look at:

- Someone on the rustlang internals list has brought up the idea of how to support explicit memory layout. https://internals.rust-lang.org/t/idea-explicit-layout-structure-definitions/5759

- There's a library for making bitfields in Rust. Although it looks a little overdone. https://github.com/dzamlo/rust-bitfield

shaneleonard, 2017-08-28:
So, it turns out that there were some limitations of @daniel-scs's implementation, and for the Teensy port, I do actually need to enforce some of the features I listed above (especially single-write, multiple field). It's not totally clean yet but I spent a lot of thought and effort on an interface which I think is better than anything I've seen (measured by the criteria I listed).

You can check it out here: https://github.com/shaneleonard/tock-teensy, feedback would be much appreciated.

The register interface itself is under common/src/regs, and I use this interface pretty much everywhere in the implementation of the drivers in chips/mk66/. For instance, register definitions are all in chips/mk66/src/regs.

The one thing I will say about my implementation is that it's pretty involved, although I've tested quite a bit and it does in fact seem to be a zero-cost abstraction. Everything that I've seen has compiled down to the optimal inlined bit twiddling instructions, there's no evidence of the abstraction if you just look at the binaries.

shaneleonard, 2017-08-28:
@brghena I've seen the bitfields library, and I don't like the fact that when reading and writing the code, you're still working with individual bits and bit ranges, so in my opinion it ends up being only slightly better than doing everything manually the C way (and less readable). The explicit memory layout thread looks promising though, I haven't seen that before, thanks for sharing!

shaneleonard, 2017-08-28:
I'll try to write up a README soon and put it in the register crate, so it's easier to quickly evaluate.

shaneleonard, 2017-08-29:
README is at https://github.com/shaneleonard/tock-teensy/blob/master/common/src/regs/README.md

daniel-scs, 2017-08-31:
Holy cow, this looks sweet.

One little thing is that it looks like the bitfields are not type-constrained to work with only their intended registers (e.g., it looks like you are accidentally reading the `CR::EN` bitfield out of the `S` register in the readme) ... but then, this is actually handy in the many cases where the same bitfield could be used on several related registers.  Maybe the docs should just make this clear.

shaneleonard, 2017-08-31:
@daniel-scs Thanks for catching my mistake! And yeah you're exactly right, there is no type constraint on setting the fields in their intended registers, with the reason being just what you described. I'll make that more clear in the doc. My intention was that by making the operations readable and well-namespaced, it should still be easy to catch those kinds of mistakes with a code review.

brghena, 2017-08-31:
Yeah, this is super interesting @shaneleonard. I really like the matches statement.

Would it be possible to make set and clear functions, such that set is just a wrapper on modify to True. For example `regs.cr.set(CR::EN);` would map to `regs.cr.modify(CR::EN::True);`. I think only working on one register field would be best, but possibly `regs.cr.set(CR::EN + CR::INT);` would be acceptable.

It feels disappointing that the bitfields and registers are separated and not checked at present. I'm not sure what a fix for that would look like though.

brghena, 2017-08-31:
Also, what's the source of the `bitfields!` macro?

shaneleonard, 2017-08-31:
The `set` and `clear` functions are technically not hard to implement, but I was worried about a naming conflict. The setter/getter functions for reading the raw register value are called `set` and `get`. Maybe `set_flag` and `clear_flag` or something similar?

shaneleonard, 2017-08-31:
The `bitfields` macro is defined in `common/src/regs/macros.rs`.

shaneleonard, 2017-08-31:
With respect to the fields being tied to specific registers/groups of registers, I think that I can also implement that, but the cost would be that it's more verbose/redundant to define the register struct. If that's acceptable, I can give it a shot. I agree that it would be a nice feature.

shaneleonard, 2017-08-31:
Yeah actually I definitely prefer `SET` and `CLEAR` over `True` and `False`. In my opinion, it'd be best to just pick one convention or the other and not have two redundant ones, but if there's a compelling reason to keep both, that's fine.

I'd love to get more people's eyes on this, so the call sounds like a great way to do that. I'm glad to hear that it would be of value to the community.

daniel-scs, 2017-10-03:
To avoid mask/shift confusion in bitfield definitions, I'd be pretty happy with `RANGE (Mask(0b11), 4)` or `RANGE Mask(0b11) 4` ... but I realize it is not quite as pretty.

ppannuto, 2018-02-28:
Merged in #658.

closed 2018-02-28 by ppannuto