tock/tock #602: Rustconf sample capsule and hail app interact to panic with full debug buffer

joshtriplett, 2017-08-18:
This happened for more than one person at Rustconf, creating the capsule that periodically printed `"Hello world!"`.

Notice that after `TOCK_DEBUG(13)`, the debug messages from the capsule stop appearing. A while after that, the panic occurs.

```
No device name specified. Using default "tock"
Using "/dev/ttyUSB0 - Hail IoT Module - TockOS"

Listening for serial output.
[Hail] Test App!
[Hail] Samples all sensors.
[Hail] Transmits name over BLE.
[Hail] Button controls LED.
[Hail Sensor Reading]
  Temperature:  2568 1/100 degrees C
  Humidity:     3934 0.01%
  Light:        15
  Acceleration: 1005
  A0:           1674 mV
  A1:           2062 mV
  A2:           2163 mV
  A3:           2346 mV
  A4:           2481 mV
  A5:           2426 mV
  D0:           0
  D1:           0
  D6:           0
  D7:           0

TOCK_DEBUG(0): src/accel.rs:20: Hello world!
[Hail Sensor Reading]
  Temperature:  2568 1/100 degrees C
  Humidity:     3934 0.01%
  Light:        15
  Acceleration: 1000
  A0:           1867 mV
  A1:           1912 mV
  A2:           1992 mV
  A3:           2108 mV
  A4:           2083 mV
  A5:           2245 mV
  D0:           0
  D1:           0
  D6:           0
  D7:           0

TOCK_DEBUG(1): src/accel.rs:20: Hello world!
[Hail Sensor Reading]
  Temperature:  2567 1/100 degrees C
  Humidity:     3931 0.01%
  Light:        15
  Acceleration: 1000
  A0:           1847 mV
  A1:           1844 mV
  A2:           1946 mV
  A3:           1927 mV
  A4:           2124 mV
  A5:           2204 mV
  D0:           0
  D1:           0
  D6:           0
  D7:           0

TOCK_DEBUG(2): src/accel.rs:20: Hello world!
[Hail Sensor Reading]
  Temperature:  2567 1/100 degrees C
  Humidity:     3929 0.01%
  Light:        15
  Acceleration: 1004
  A0:           1718 mV
[... 514 lines cut ...]
  0x00030058 ┼─────────────────────────────────────────── A
             │ Protected        88                        S
  0x00030000 ┴─────────────────────────────────────────── H

  R0 : 0x00000000    R6 : 0x20004BA0
  R1 : 0x20004790    R7 : 0x000007F8
  R2 : 0x200047D6    R8 : 0x000008B9
  R3 : 0x00000002    R10: 0x00000891
  R4 : 0x00000003    R11: 0x0000000F
  R5 : 0x200047D6    R12: 0x000007F8
  R9 : 0x20004800 (Static Base Register)
  SP : 0x20004768 (Process Stack Pointer)
  LR : 0x0003074D
  PC : 0x20004790
 YPC : 0x0003072E

 APSR: N 0 Z 0 C 0 V 0 Q 0
       GE 0 0 1 1
 IPSR: Exception Type - IRQn
 EPSR: ICI.IT 0x02
       ThumbBit false !!ERROR - Cortex M Thumb only!
 To debug, run `make debug RAM_START=0x20004000 FLASH_INIT=0x300b1`
 in the app's folder and open the .lst file.

```

bradjc, 2017-09-21:
Ok, after a little searching this seems to stem from the fact that kernel "apps" (like the `debug!()` module) are not enough like real apps (actual running processes). In particular, when we iterate through apps to find if there is any outstanding work to be done, that iteration does not cover the kernel "apps".

So this:

```rust
for cntr in self.apps.iter() {
    ...
}
```

from here

https://github.com/helena-project/tock/blob/afa1a9e8db3a5de1e64ac9a965c26e1f21618c2a/capsules/src/console.rs#L280-L281

will only ever search through the actual processes. The issue is that if `debug!()` tries to write to the console but the console is busy, the console driver will queue the request, but then we never check for those queued requests because that loop doesn't iterate the kernel "apps".

This stems from this setup where the length is only set to the number of processes, but simply increasing that runs into other issues in that file:

https://github.com/helena-project/tock/blob/afa1a9e8db3a5de1e64ac9a965c26e1f21618c2a/kernel/src/container.rs#L246-L255

To get this working I came up with following diff. It's a hack, but I think it proves that this is the issue.

```diff
diff --git a/capsules/src/console.rs b/capsules/src/console.rs
index 3d8fe20b..8ecfecaa 100644
--- a/capsules/src/console.rs
+++ b/capsules/src/console.rs
@@ -278,8 +278,10 @@ impl<'a, U: UART> Client for Console<'a, U> {
         // If we are not printing more from the current AppSlice,
         // see if any other applications have pending messages.
         if self.in_progress.get().is_none() {
+            let mut found_one = false;
             for cntr in self.apps.iter() {
                 let started_tx = cntr.enter(|app, _| {
+
                     if app.pending_write {
                         app.pending_write = false;
                         match self.send_continue(app.appid(), app) {
@@ -299,9 +301,40 @@ impl<'a, U: UART> Client for Console<'a, U> {
                     }
                 });
                 if started_tx {
+                    found_one = true;
                     break;
                 }
             }
+
+            if found_one == false {
+                // check the kernel
+
+                self.apps.enter(AppId::new(255), |app, _| {
+                    if app.pending_write {
+                        app.pending_write = false;
+                        let y = match self.send_continue(app.appid(), app) {
+                            Ok(more_to_send) => more_to_send,
+                            Err(return_code) => {
+                                // XXX This shouldn't ever happen?
+                                app.write_len = 0;
+                                app.write_remaining = 0;
+                                app.pending_write = false;
+                                let r0 = isize::from(return_code) as usize;
+                                app.write_callback.map(|mut cb| { cb.schedule(r0, 0, 0); });
+                                false
+                            }
+                        };
+                    }
+                }).unwrap_or_else(|err| {
+                   let k =  match err {
+                        Error::OutOfMemory => ReturnCode::ENOMEM,
+                        Error::AddressOutOfBounds => ReturnCode::EINVAL,
+                        Error::NoSuchApp => ReturnCode::EINVAL,
+                    };
+                });
+
+
+            }
         }
     }
 
```

alevy, 2017-10-09:
I'm removing this from the 1.0 Milestone, since (a) we just won't finish fixing this before tomorrow, and (b) it doesn't seem to have any impact on the system call interface, which is what 1.0 is stabilizing. Please holler if you protest.

commit a2003473 (2018-06-12): kernel: grant: add debug app to iterator

fixes #602