tock/tock #617: Radio Race Condition

ptcrews, 2017-09-16:
Hi all,

There seems to be a race condition in the RF233 when receiving frames. This bug can be reliably replicated by running the `lowpan_frag_dummy.rs` test suite (located on the branch `ptcrews/fragment_pr` and included in pull request #581). Tests number 7 and 25 fail, where the middle frame is corrupted on reception. Using Wireshark, we are able to confirm that each frame is sent correctly, as the full packet can be reassembled and decoded by Wireshark. By inserting a short delay before sending each fragment, we were able to eliminate the problem. This indicates that the bug is a result of incorrectly handling the reception of multiple frames.

I intend to update this post with more information as I do more testing, but wanted to document the issue in case anyone else runs into it, or if I am unable to fix it.

phil-levis, 2017-09-16:
Just to follow up, IIRC the best hypothesis so far is that a frame starts arriving while one is being read out, and so it corrupts the end of the being-read-frame. One fix needed there would be to disable reception while a packet is being read out; another is to increase the SPI bus speed so it can be read out faster.

Phil

alevy, 2017-09-26:
Disabling reception while a frame is being read out seems like the better solution. Speeding up SPI might help mitigate this, but it won't eliminate the problem entirely.

Though I maybe we hope that CRC or a MAC somewhere in a higher layer would catch it anyway, and it'll be rare with a high SPI bus speed?

alevy, 2017-09-26:
OK, so the RF233 seems to have the ability to prevent overwrites in hardware by turning on "Dynamic Frame Buffer Protection" (section 11.8 of the RF233 datasheet). As far as I understand, since we're setting `RX_SAFE_MODE` in the `TRX_CTRL_2` register, a new frame shouldn't overwrite the existing one until `/SEL = H`.

What I believe might be happening is that when an IRQ occurs (in this case indicating the `TRX_END` event), our current implementation reads the `IRQ_STATUS` register. _This_ operation completes and releases the virtual SPI device, which will release the select line.

I know what you're thinking, @phil-levis: power locks

There _is_ technically a way around that, I think. We don't actually need to read the `IRQ_STATUS` register until we're done reading the packet (and then only to clear the interrupt). The events are completely governed by the state machine (see figures 7-1 and 7-11 in the datasheet). We know, based on our previous state, what the IRQ means.

hudson-ayers, 2018-02-24:
Just to update this here, I just ran into this problem again in the 6lowpan test suite after updating the test suite to work with the new network stack design. For now I got around it by inserting a 1s delay between transmitting individual frames within a packet.

phil-levis, 2018-02-24:
I'm going to put this at the top of my queue and start working on it Monday morning.

phil-levis, 2018-02-26:
Yeah, 11.8 (Dynamic Frame Buffer Protection) is totally insufficient -- not only do we read the IRQ status register, a frame read is two separate SPI operations, one to read length and one to read the whole frame. Since the RF233 requires that /SEL be toggled between operations, the hardware basically requires that you do byte-by-byte SPI operations, so as you point out, you have to lock it. We weren't crazy....

Beyond fully implementing power locks for the SPI, I think the right fix is to disable reception while a frame is being read out (so doing Dynamic Frame Buffer Protection in software).

bradjc, 2018-03-05:
I think this has been taken care of as of #758

closed 2018-03-05 by bradjc