tock/tock #618: Incomplete SysTick implementation

shaneleonard, 2017-09-18:
In the `SysTick::value` function, if the TENMS field calibration register is zero, this leads to a panic caused by trying to divide-by-zero. This hasn't been a problem before because the TENMS field is nonzero on the SAM4L. However, on the Teensy's Freescale processor, it is zero, so this unhandled case makes the SysTick unusable on the Teensy.

The problematic code (`arch/cortex-m4/src/systick.rs`):

```rust
    fn value(&self) -> u32 {
        let tenms = self.calibration.get() & 0xffffff;
        let value = self.value.get() & 0xffffff;

        value * 10000 / tenms
    }
```

We see that if `tenms` is zero, this triggers a panic. The [ARM Cortex-M4 manual entry](http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dui0553a/Bhcjegci.html) for the calibration register describes the TENMS field as follows:

> Reload value for 10ms (100Hz) timing, subject to system clock skew errors. If the value reads as zero, the calibration value is not known.

In the case where TENMS is zero:

> If calibration information is not known, calculate the calibration value required from the frequency of the processor clock or external clock.

Thus, the correct implementation would check TENMS and calculate it if necessary.

alevy, 2017-09-19:
Oh thanks! Some thoughts:

  * One problem with implementing calibration somehow generically is that it requires some external clock, which is chip/platform dependent, rather than architecture dependent, right? One option is to add a `calibrate` function which defaults to a no-op function, panics if TENMS is zero or maybe just sets an arbitrary value (since we only use systick for time slicing, it's probably not a big deal if the the reload time isn't really 10ms).

  * TENMS is not writable, which is a bummer because it means we need to keep some state outside the registers for calibration...

ppannuto, 2018-01-21:
Did #713 address all the problems here?

alevy, 2018-01-22:
Yes!

closed 2018-01-22 by alevy