tock/tock #625: Unused chip drivers still take up RAM

brghena, 2017-10-01:
I was looking through RAM use on Signpost today, and wanted to mark this down while I was thinking about it. It's not a blocking problem in any way, but seems like undesirable behavior.

On Signpost (the Audio Module specifically) the USB client driver for the SAM4L takes up 288 Bytes of RAM even though it isn't in use and isn't included in the board's `main.rs` file. The RAM is spent on the USBC struct (allocated as `pub static mut` [here](https://github.com/helena-project/tock/blob/master/chips/sam4l/src/usbc/mod.rs#L870)).

My guess as to why it's taking space at all is because it's included in the `chip.rs` and `mod.rs` files in order to connect it to the interrupt handlers. This feels generally undesirable.

alevy, 2017-10-01:
> My guess as to why it's taking space at all is because it's included in the chip.rs and mod.rs files in order to connect it to the interrupt handlers. This feels generally undesirable.

It's indeed exactly that. I have changes (long overdue for master) I'm getting ready to make into a PR that change how we handle interrupts (as discussed on the mailing list previously), which will get us part of the way there (removing the ISRs array), but will not fix the issue in `chip.rs`'s `handle_interrup` dispatch code.

This seems like a job for some sort of macro, but i'm not totally sure how to do it.

enzuru, 2018-01-29:
@alevy is there any issue with me trying my hand at this ticket?

alevy, 2018-01-29:
@enzuru of course not!

hudson-ayers, 2020-10-08:
This is addressed by the design in #2069, but needs to be applied to every board/chip combo before this can be closed.

hudson-ayers, 2021-01-02:
Closed by #2069 and the other related PRs. 

For an example of how to create a minimal board that does not use all chip drivers, see this PR: https://github.com/tock/tock-bootloader/pull/19

closed 2021-01-02 by hudson-ayers