tock/tock #630: Stabilize GPIO system call ABI

alevy, 2017-10-02:
As discussed briefly in #559 (and a bit more on chat), there are some questions remaining about the GPIO interface.

Some questions that came up:

- Do we need a `make_output` call, or should it simply be implicit in `set`/`clear`/`toggle`?
- What should the semantics be  if interrupts are enabled but the pin is in output mode?

@bradjc (and others) please add to the question list as you see fit by just editing the main text (rather than in a comment, so we have it in one place).

bradjc, 2020-01-29:
Did we implicitly do this by not doing anything?

bradjc, 2020-04-13:
My opinion at the time was: GPIO is so common, and so often a starting point for developers, and since we can have an in-kernel interface (“advanced”) that is different from the userspace interface (“user friendly”), we should take advantage of that to make a userspace API that is easy to use and works for 98% of use cases. In the meantime, however, we have had the more "advanced" userspace API, and we haven't had any issues. Plus, what maybe wasn't clear to me at the time, but is now pretty obvious is: basically no one actually looks at the syscall interface, and instead just use the libtock libraries, and it makes more sense to hide the complexity there.

Result: we should stabilize the interface as-is.

ppannuto, 2020-04-21:
Closed by #1759.

closed 2020-04-21 by ppannuto