tock/tock #643: Wishlist [publish tock on crates.io after stable rust userland]

niklasad1, 2017-10-07:
Hi,
After rust in userland has been stabilized it would be very very cool to publish tock on [crates.io](https://crates.io/).
Then people could easily load tock easily by something like:
```rust
extern crate tock_board_XX
extern crate libtock

fn main() {
  loop {
    let _ = libtock::led_on(1);
    let temp = libtock::read_temperature(1)
    write!("temperature {}", temp);
  }
}
```
Well, technically the kernel must be built and loaded first and then user-space processes.

Under the hood the following is built:
1. the kernel for a specific board (e.g. make TOCK_BOARD=xx)
2. libtock & user app

Two binaries that the need to be flashed to the board.

Is this technically possible? (I haven't checked exactly how cargo/xargo works but seems cool)

bradjc, 2018-05-08:
Should we move this to libtock-rs?

Do we want to publish the myriad of tock kernel crates to crates.io?

bradjc, 2018-11-29:
This hasn't come up in over a year, so not much interest at the moment. We have pushed things in `/libraries` to crates.io by request, so it seems that the code people are interested in easy access to is getting out there.

closed 2018-11-29 by bradjc