tock/tock #644: hil::uart issues

phil-levis, 2017-10-09:
Going through the process of implementing a virtualizer for the UART HIL, I've come across a couple of interface problems.

1) The UART trait combines the control path (init) with the data path (transmit/receive); this implies that one cannot virtualize, unless they somehow alternate configurations (e.g., what if one client specifes 8-N-1 and another specifies 7-parity-2?

2) The transmit and receive paths are combined. This means, for example, that printf also needs to handle receive events. These should be separated into separate traits.

3) The calls to transmit and receive (including advanced ones) do not return ReturnCode so cannot fail (or push failures to the callbacks). The callbacks use a different Error type which is very UART-specific.

bradjc, 2018-06-28:
>The UART trait combines the control path (init) with the data path (transmit/receive); this implies that one cannot virtualize, unless they somehow alternate configurations (e.g., what if one client specifes 8-N-1 and another specifies 7-parity-2)?

That's almost by design. UART busses aren't really shared busses, and for all the chips/boards we have I don't think this has been an issue.

>The transmit and receive paths are combined. This means, for example, that printf also needs to handle receive events. These should be separated into separate traits. This also prevents effective virtualization: transmission virtualization involves queueing, while reception virtualization involves dispatch.

This could be useful. I wonder if the console capsule would be different if the traits were separated.

> The calls to transmit and receive (including advanced ones) do not return ReturnCode so cannot fail (or push failures to the callbacks).

This should be fixed. The UART trait likely predates the ReturnCode type.

> The traits still explicitly use 'static for Client references rather than be generic types with a <'a> parameter.

Arguably `'static` is the standard since the vast majority of HIL interfaces use it. I tried to see if changing it would be super simple, but it does require changes outside of the HIL folder.

bradjc, 2019-06-04:
I think with the updates to the UART HIL this has been addressed.

closed 2019-06-04 by bradjc