tock/tock #647: Following the tutorials to the letter does not work with the Imix board

hudson-ayers, 2017-10-10:
As a continuation of the issues with the tutorial/Getting Started Guide/Imix README that I mentioned last night on basecamp, I met with Shane today and figured out how to get tutorial02 (button_print) working on the Imix v2.0.  This brought several issues to light.

**Explanation of the problems:**

1. The Getting Started Guide recommends installing tockloader v0.7.1, but this results in a number of warnings that recommend updating tockloader.

2. The Getting Started Guide instructs users to reference the board specific READMEs to program apps. The Imix guide, however, simply instructs users to upload applications using `make program` . Unfortunately, if a user calls make program from any of the example application directories without first changing `TOCK_BAORD ?= hail` to `TOCK_BOARD ?= imix` in AppMakefile.mk then this call will overwrite a portion of the kernel by trying to load an application into the location that would be correct for the hail board. Alternatively, `make TOCK_BOARD=imix program` will program apps correctly

3. The imix guide also states:
"This builds and loads only a single app. Tock is capable of running multiple apps concurrently. Use `tockloader install` to add additional apps, and `tockloader list` to see the list of installed applications."

Unfortunately, as a result of the 6lowpan pull request being merged, the location of the application addresses are now different than tockloader expects. As a result, using `tockloader install` to load an application onto the board will report success, but will actually be overwriting part of the kernel, and will not result in a functional application. Using `tockloader list` will report that the application resides on the board, despite the fact that this application will not run.

Currently, in order for tockloader to work effectively on the Imix, all tockloader commands that must know the address of the application space must be appended with `-a 0x40000` Ex: `tockloader list -a 0x40000`. This information is not detailed anywhere in the Imix README, getting started guide, or tutorials.

**Proposed Solutions:**

1. Either update the Getting Started guide to have users install tockloader v0.7.2 or update the guide to inform users these warnings will appear and that they are not of concern.

2. Update the Imix README to instruct users to make the necessary changes to tock/userland/AppMakefile.mk

3. Shane says that he is working on an update that will add an attribute which specifies the application address location so that tockloader can automatically query this attribute rather than require users to figure out where this location is and pass it to tockloader manually. 

In my opinion, an ideal solution would also call `tockloader list-attributes` and automatically update the TOCK_BOARD field in  AppMakefile.mk based off of the response to this command. I realize, however, that this may not be as easy as it sounds. 

Depending on how long Shane's fix will take, the tutorials/Getting Started Guide should possibly be updated to reflect the necessary steps to use tockloader (and make program) correctly on the Imix - especially if Shane's fix will not be ready by the time of any demonstrations on Imix boards at conferences.

alevy, 2017-10-11:
> update the Getting Started guide to have users install tockloader v0.7.2 

Probably makes sense

> Update the Imix README to instruct users to make the necessary changes to tock/userland/AppMakefile.mk

I think the change we want at the moment is to instruct users to install apps like this:

```
$ make TOCK_BOARD=imix program
```

> In my opinion, an ideal solution would also call tockloader list-attributes and automatically update the TOCK_BOARD field in AppMakefile.mk based off of the response to this command. I realize, however, that this may not be as easy as it sounds.

In general, we don't want to make changes to _code_ (a Makefile is sort of code) to address "runtime" (in this case running the programming utility) concerns. So, we definitely want to use bootloader attributes to figure out which board we're using or where the apps live, but we don't want to modify `AppMakefile.mk`, etc. Shane's fix should make it relatively straightforward to do the former, though, so I think we're good.

bradjc, 2017-10-11:
#367 (kernel attributes) was designed to address this. We can put the app location in flash in the bootloader/board attributes, but those are designed to be board-specific and not kernel-specific. One particular problem is that the address has to be kept in sync between the kernel and that attribute, which is awkward when the attribute is in the bootloader.

In the short-term any of the solutions posed are great, I just want to add in what is likely the longer-term solution.

bradjc, 2018-05-03:
I think we have fixed this.

closed 2018-05-03 by bradjc