tock/tock #659: [nRF5X Bluetooth Low Energy] Don't work on iPhone

niklasad1, 2017-10-17:
Hi,

As noticed in [#649](https://github.com/helena-project/tock/pull/649) the Bluetooth Low Energy advertisements can't be found on iPhone using the app  ```nRF Connect``` and need to be fixed!

Currently, it works on Android (at least on Samsung S6).
It need to be investigated further!!

niklasad1, 2017-11-29:
Well, since I updated iOS it seems to work fine anyone who can verify?

alevy, 2017-11-29:
It would be useful to figure out if this is a bug in iOS, in our advertising driver implementation, or in the particular advertising packet the test app sends. E.g. seeing if an advertisement generated from the nRF-SDK shows up.

I don't have an iOS device so I can't help and it sounds like at @niklasad1's now upgraded device won't help us anymore either (since it works now) :).

I'm inclined to close this for now. Thoughts?

bradjc, 2017-11-29:
We can always re-open it if anyone can reproduce it.

ppannuto, 2017-11-29:
I can test this afternoon when I get in and close if it’s fixed here too

ppannuto, 2017-11-29:
Can confirm that this still an issue and the advertisement does not appear on iOS 11.1.2 with nrfConnect or LightBlue. Branden found it on his phone, but the MAC address is listed as `ANONYMOUS`, which we've never seen before:

[image]

alevy, 2017-11-29:
@ppannuto this has not generally been an issue you for you all with  the NRFs, right? I.e. you have a bunch of applications that use the NRF51 with the Nordic softdevice---I think you've never had this kind of issue?

niklasad1, 2017-11-29:
Thanks, false positive I must have tested this on my other branch sorry :cry: 
Been a while since I been on the real ```master branch``` looks pretty weird on Android too my. 
The app is currently using address [0, 0, 0, 0, 0, 0] which may be the problem since it is not valid BLE address.

However, this works smooth on my other branch so this issue will remain open until that one is merged!

ppannuto, 2017-11-29:
I don't recall it being an issue before, but I also haven't tested nrf BLE much in a while, a bunch has changed. I suspect the invalid BLE address is the issue - iOS is presumably just dropping it.

brghena, 2017-11-29:
OH. Yeah, having all zeros as the address probably makes it not show up on iOS. I'd believe that easily.

PR #701 (2017-12-02): Use valid BLE advertisement address for NRF5X
### Pull Request Overview

This pull request fixes so that NRF5X is using valid Bluetooth Low Energy address during advertising to appear on both Android and iOS.

Also closes #659 

### Testing Strategy
```ble advertising app``` was tested on both nRF51-DK and nRF52-DK

### TODO or Help Wanted

### Documentation Updated

~~- [ ] Kernel: The relevant files in `/docs` have been updated or no updates are required.~~
~~- [ ] Userland: The application README has been added, updated, or no updates are required.~~

### Formatting

- [x] `make formatall`  [... cut]
closed 2017-12-02 by alevy