tock/tock #676: nRF5x crate limits nRF52 to 4 GPIOTE channels

JayKickliter, 2017-11-01:
nRF51 has 4 interrupt channels, but nRF52 has 8. So the nRF5x crate is falsely limiting nRF52 to 4.

The result is that when I try to enable more than 4 GPIO interrupts, I get a kernel panic in debug mode:

```
Kernel panic at vendor/tock/chips/nrf5x/src/gpio.rs:172:
        "attempt to shift left with overflow"
```

niklasad1, 2017-11-01:
Ok, we need fix that!

Shall we pull out the entire GPIO driver from the nRF5X crate and define separate GPIO drivers in nRF51 and nRF52?

Or keep the current driver in nRF5X and complement with additional functionality in the nRF52 crate?

JayKickliter, 2017-11-01:
Those were my questions. It could be a feature flag. I don't really have an opinion on it, but I'm going to have to fork Tock to get it working quickly.

PR #677 (2017-11-04): Generalize nrf5x GPIOTE to work more channels
### Pull Request Overview

The NRF51 and NRF52 have a different number of GPIOTE channels (four and
eight, respectively). This pull request generalizes the GPIOTE structure to
accomodate both using a const that's feature gated on either the `nrf51` or `nrf521` feature.

To choose between them, the depending chip crate uses:

```
[dependencies.nrf5x]
path = "../nrf5x"
features = ["nrf51"]
```

Closes #676 

### Testing Strategy

_TODO_

### TODO or Help Wanted

@niklasad1 or @JayKickliter might be able to help test this (I'm out of reach of my nrf5' [... cut]
closed 2017-11-04 by alevy