tock/tock #715: Support Cortex-M3

Niondir, 2018-01-02:
We are considering to use Tock on our Cortex-M3 based hardware. I would love to see a port. Is there anything in the roadmap?

niklasad1, 2018-01-02:
Hi @Niondir,
There is a port see [https://github.com/helena-project/tock/pull/686](https://github.com/helena-project/tock/pull/686) but I don't know whether it will go into the Tock repo or not but it is functional what I have understood.

alevy, 2018-01-02:
Which Cortex-M3, @Niondir?

Niondir, 2018-01-04:
We are using the STM32L151CB-A

alevy, 2018-01-05:
Yeah, that looks like a perfect chip for a Tock port. Is there a reasonably available development board for it? E.g. a discovery board or nucleo that developers would be able to use?

As @niklasad1 said, there is an active effort to port to cortex-m3 and stm in particular, but the current port is using a slightly less than ideal chip (it doesn't have an MPU). It's definitely on the roadmap to either merge that and keeping working on it, move it to a different repo and keep working on it, or find a more suitable board first and work on that. And we'd like to do that ASAP, we're well aware that STM is a popular family of chips.

george-hopkins, 2018-01-25:
In the meantime, the ports for STM32 were moved to (tock/tock-stm32)[https://github.com/tock/tock-stm32]. We already have a port for STM32F1 and one for STM32F4 soon. If you'd like to start a new port, feel free to contact us over there, on the mailing list or on IRC.

alevy, 2018-01-25:
Closing this for now since, as @george-hopkins noticed this is kind of resolved by the tock/tock-stm32 repo. Though It raises the point that eventually we'll want to upstream at least the `cortex-m3` architecture crate from that repo.

closed 2018-01-25 by alevy

Niondir, 2018-01-29:
I guess it's close enough to STM32L151CB-A to create another port based on on the fork. Currently I'm too busy, but maybe one day when I find some spare time.