tock/tock #716: [RFC] Blutooth Low Energy Advertising System-Call Interface

alevy, 2018-01-05:
- Feature Name: process-control-of-ble-advdata
- Start Date: 2018-0104

# Summary
[summary]: #summary

A BLE advertisement packet looks like this:

```
+----------+--------+-----------+
|  Header  |  AdvA  |  AdvData  |
+----------+--------+-----------+
```

Where:

  * `Header` specifies the packet _type_ (e.g. directed, undirected, connectable, etc), length, a `TxAdd` bit which specifies if `AdvA` is a public or random address, and an `RxAdd` bit which specifies if `AdvData` contains a public or random address when the PDU type is a directed advertisement (`ADV_DIRECT_IND`) and otherwise is unused.

  * `AdvA` is the advertiser's public or random device address

  * `AdvData` is the actual data being sent. If the PDU type is a directed advertisement, `AdvData` is the target device address. For all other advertisements it is an arbitrary applicatoin payload. For GAP applications (most typical), it is defined by the Core Specification Supplement.

The recently merged (#661) BLE advertising driver exposes a system call interface to processes that only allows the process to control fields of a GAP packet. Instead, this proposal would let userspace processes set the raw `AdvData` portion of an advertising packet. That packet _could_ contain GAP data, perhaps with the help of a userspace library, or could could use other forms of data.

# Motivation
[motivation]: #motivation

The most goal of the system call interface design is to maximize userspace flexibility under reasonable performance constraints and subject to isolation between processes.

We can achieve the same level of isolation between processes, where each process has complete control over a unique device address, without forcing the GAP applicatoin protocol. This is advantageous because it gives processes more flexibility (e.g. to not use GAP), it reduces the complexity in the kernel driver since it doesn't need to know anything about the GAP protocol, and it simplifies the system call interface significantly. It also looks a lot closer to the BLE HCI protocol so, in principle, could make it simpler to port the same system call interface for use with an HCI peripheral. 

Even for applications that _use_ GAP, having this flexibility can be advantageous. For example, updating the advertising packet may only require updating a few bytes in place rather than reforming the entire packet, but it is not possible to do that with the current interface.

# Guide-level explanation
[guide-level-explanation]: #guide-level-explanation

A BLE advertisement packet looks like this:

```
+----------+--------+-----------+
|  Header  |  AdvA  |  AdvData  |
+----------+--------+-----------+
```

The `Header` and `AdvA` fields are controlled by the kernel to reflect the process's advertising addres, the packet _type_ and the , while the `AdvData` field is set by the process.

To send an advertising packet, the process first  sets advertisment parameters such as the tx_power, advertisement type, interval, etc using their respective `command` system calls. It then sets the `AdvData` field by passing a payload up to 31 bytes to the `set_advdata` allow system call.

The `set_advdata` system call copies the data into its own buffer immediately and releases the buffer back to the process when the call completes.

Finally, the process calls the `start_advertising` system call.

To update the `AdvData` field when advertising is enabled, the process simply calls `set_advdata` again with the same or another buffer. The driver will use this new buffer in the following advertising event.

# Reference-level explanation
[reference-level-explanation]: #reference-level-explanation

The change would be relatively straight forward:

Replace the current allow for [`BLEGap`](https://github.com/helena-project/tock/blob/3a98489ed7980076c3867e526f6030fc42ec60ab/chips/nrf5x/src/ble_advertising_driver.rs#L919) with one that simply takes the provided `AppSlice` and copies as-is into the `AdvData` portion of the `app_write` field.

As a result, we would be able to get rid of a significant amount of GAP-related code in the driver.

# Drawbacks
[drawbacks]: #drawbacks

This forces the each process to contain the complexity of the GAP protocol. In practice, though, most applications will be much simpler than all of GAP and will be able to use only a subset.

# Rationale and alternatives
[alternatives]: #alternatives

  * One alternative would be to simply leave the interface as is. This is suboptimal because of the reasons listed in the [motivation] section.

  * Another option would be to support both interfaces, but that would retain all of the complexity of GAP in the driver with no clear benefit.

  * Instead of the `allow` system call releasing the buffer upon return, it could retain access to it. In cases where the process keeps this buffer around anyway, it would allow the driver to save one copy of the buffer. However, it would require some form of synchronization in order to allow the process to modify the buffer safely. For example, the kernel might interrupt a process in the middle of modifying the buffer exactly to copy the buffer into the low level radio driver to send the next advertising event.

# Unresolved questions
[unresolved]: #unresolved-questions

- Are there ways to break isolation by completely controll `AdvData`?

alevy, 2018-01-05:
/cc @niklasad1

alevy, 2018-01-05:
Side-note: I really feel that we need an RFC repo, like Rust does for these kinds of things...

niklasad1, 2018-01-05:
I don't want to be too annoying but the actual payload differ between different advertising/scanning PDUs but since we only support `NON_CONNECT_IND` that is not so important at the moment anyways.

But I write this up because I'd just added some inline comments about this and it might be useful!

 **Packet format**
```
  +--------+---------+
  | Header | Payload |
  +--------+---------+

```
 **Advertising/Scanning PDUs**
```rust
// ConnectUndirected (ADV_IND): connectable undirected advertising event
// BLUETOOTH SPECIFICATION Version 4.2 [Vol 6, Part B], section 2.3.1.1
//
//   PDU     +-----------+      +--------------+
//           | AdvA      |  -   | AdvData      |
//           | (6 bytes) |      | (0-31 bytes) |
//           +-----------+      +--------------+
//
// ConnectDirected (ADV_DIRECT_IND): connectable directed advertising event
// BLUETOOTH SPECIFICATION Version 4.2 [Vol 6, Part B], section 2.3.1.2
//
//   PDU     +-----------+      +--------------+
//           | AdvA      |  -   | InitA        |
//           | (6 bytes) |      | (6 bytes)    |
//           +-----------+      +--------------+
//
// NonConnectDirected (ADV_NONCONN_IND): non-connectable undirected advertising event
// BLUETOOTH SPECIFICATION Version 4.2 [Vol 6, Part B], section 2.3.1.3
//
//   PDU     +-----------+      +--------------+
//           | AdvA      |  -   | AdvData      |
//           | (6 bytes) |      | (0-31 bytes) |
//           +-----------+      +--------------+
//
//
// ScanUndirected (ADV_NONCONN_IND): scannable undirected advertising event
// BLUETOOTH SPECIFICATION Version 4.2 [Vol 6, Part B], section 2.3.1.4
//
//   PDU     +-----------+      +--------------+
//           | AdvA      |  -   | AdvData      |
//           | (6 bytes) |      | (0-31 bytes) |
//           +-----------+      +--------------+
//
// ScanRequest (SCAN_REQ)
// BLUETOOTH SPECIFICATION Version 4.2 [Vol 6, Part B], section 2.3.2.1
//
//   PDU     +-----------+      +--------------+
//           | ScanA     |  -   | AdvA        |
//           | (6 bytes) |      | (6 bytes) |
//           +-----------+      +--------------+
//
// ScanResponse (SCAN_RSP)
// BLUETOOTH SPECIFICATION Version 4.2 [Vol 6, Part B], section 2.3.2.2
//
//   PDU     +-----------+      +--------------+
//           | AdvA      |  -   | ScanRspData  |
//           | (6 bytes) |      | (0-31 bytes) |
//           +-----------+      +--------------+
//
// ConnectRequest (CON_REQ)
// BLUETOOTH SPECIFICATION Version 4.2 [Vol 6, Part B], section 2.3.3.1
//
//   PDU     +-----------+      +--------------+     +--------------+
//           | InitA     |  -   | AdvA         |  -  | LLData       |
//           | (6 bytes) |      | 6 bytes      |     | 22 bytes     |
//           +-----------+      +--------------+     +--------------+
//

```

niklasad1, 2018-01-05:
Regarding moving management of gap data I'm not sure if I understand it correctly,  you suggest to construct a `finalized advertisement buffer` in user-space i.e., remove  [`advertisement_buf`](https://github.com/helena-project/tock/blob/3a98489ed7980076c3867e526f6030fc42ec60ab/chips/nrf5x/src/ble_advertising_driver.rs#L337) from the Grant and keep only configuration data (tx_power, advertisement interval, alarm)?

Does that mean that user-space manages generating AdvA, decides Advertising PDU and similar?

> Even for applications that use GAP, having this flexibility can be advantageous. For example, updating the advertising packet may only require updating a few bytes in place rather than reforming the entire packet, but it is not possible to do that with the current interface.

Please elaborate, I don't understand this! Because, we don't need to re-construct the entire packet  for example upon a `GAP Type` we only append it to the advertisement buffer.

However, an additional disadvantage with the current implementation is inefficient use of system calls e.g., if the a process formats the advertising buffer completely only one `allow call` is needed!

I'm a little bit concerned to give such responsibility to user-space because we will lose `memory/type-safety`.

alevy, 2018-01-05:
> I don't want to be too annoying but the actual payload differ between different advertising/scanning PDUs but since we only support NON_CONNECT_IND that is not so important at the moment anyways.

Right, this is referring _specifically_ to sending Advertising PDUs, not scanning on initiating PDUs. That's one reason the suggestion doesn't let the process control the header: so the driver can control which PDUs payloads the process has raw access to. Currently the driver only supports NON_CONNECT_IND, but this RFC works for all advertising PDUs except ADV_DIRECT_IND (which doesn't really have an application payload).

> do you suggest to construct a finalized advertisement buffer in user-space i.e., remove advertisement_buf from the Grant and keep only configuration data (tx_power, advertisement interval, alarm) in the Grant?

No, I'm suggesting the application pass a complete `AdvData` buffer, and that will be copied into the `AdvData` portion of the `advertisement_buf`. Alternatively, we could keep separate `address`, `pdu_type` and `adv_data` fields in the `App` struct, and construct the packet when we copy it into the `kernel_buf`. I don't think there is a meaningful semantic difference.

> Please elaborate, I don't understand this! Because, we don't need to re-construct the entire packet for example upon a GAP Type we only append it to the advertisement buffer.

Ah, except that doesn't actually work if you want to update an existing field. For example, if I had an update to a temperature reading, currently, I would have to reset the payload and add all fields again. 

_If_ we wanted to support in-place updates in the current interface we would either have to parse the existing payload in `advertisement_buf`, find the field and update it, potentially rearranging the payload if the size of the field has changed as well, _or_ maintain a structured version of the GAP fields in `App` and serialize on demand. Either way, a lot of complexity would need to be added to the kernel driver to properly support updates.

Of course, such an implementation would still be inflexible to support either non-GAP payloads or extensions to GAP-payloads.

> I'm a little bit concerned to give such responsibility to user-space because we will lose memory/type-safety.

How is that we would lose memory/type-safety?

niklasad1, 2018-01-05:
> Right, this RFC is referring specifically to sending Advertising PDUs, not scanning on initiating PDUs. That's one reason the suggestion doesn't let the process control the header: so the driver can control which PDUs payloads the process has raw access to. Currently the driver only supports NON_CONNECT_IND, but this RFC works for all advertising PDUs except ADV_DIRECT_IND (which doesn't really have an application payload).

Sounds good to me

> No, I'm suggesting the application pass a complete AdvData buffer, and that will be copied into the AdvData portion of the advertisement_buf. Alternatively, we could keep separate address, pdu_type [...]

Yeah, that sounds like good idea actually which means that it is fairly easy to update the header (length of the packet) and don't have to keep track of the state of AdvData

> Ah, except that doesn't actually work if you want to update an existing field. For example, if I had an update to a temperature reading, currently, I would have to reset the payload and add all fields again.
If we wanted to support in-place updates in the current interface we would either have to parse the existing payload in advertisement_buf, find the field and update it, potentially rearranging the payload if the size of the field has changed as well, or maintain a structured version of the GAP fields in App and serialize on demand. Either way, a lot of complexity would need to be added to the kernel driver to properly support updates.
Of course, such an implementation would still be inflexible to support either non-GAP payloads or extensions to GAP-payloads.

Right right, yes this makes sense to me and the implementation becomes simpler. However, in practice this re-construct the entire AdvData every time though but it far better than parsing the packet IMO.

> How is that we would lose memory/type-safety?

I refer to that apps/libtock is written in C

Overall, this sounds good to me and should not be that hard to implement

alevy, 2018-01-05:
> I refer to that apps/libtock is written in C

Oh I see. Well we should fix that too 😛

alevy, 2018-03-28:
@niklasad1 I forgot about this RFC, I meant to incorporate this discussion into the design doc in #838