tock/tock #726: tools/nRF51_codegen.py doesn't work

fhars, 2018-01-15:
It was merge conflict resolved into an unusable state in:

commit e153968d6d0373cfede6a21d1524255d0d858d21 (HEAD)
Merge: fa780a54 3e71eae4
Author: Philip Levis <pal@cs.stanford.edu>
Date:   Fri Aug 19 01:24:40 2016 -0400

    Merge remote-tracking branch 'origin' into nRF51822/devel

Symptoms:

Traceback (most recent call last):
  File "tools/nRF51_codegen.py", line 128, in <module>
    main()
  File "tools/nRF51_codegen.py", line 118, in main
    dump_macros(interrupts,
NameError: global name 'dump_macros' is not defined

bradjc, 2018-01-23:
Are you looking to use `nRF51_codegen.py` for something in particular? I don't think it has been used for Tock in a long while, and I'm not really sure why it is still in the repo. This came up a little while ago, and a couple minor changes makes the python work, but I'm not sure how useful the output is anymore.

fhars, 2018-01-23:
No, I just noticed that it is there and was not quite sure if it might have been useful if i worked.

niklasad1, 2018-01-23:
@fhars 
I think [svd2rust crate](https://crates.io/crates/svd2rust) could be what you are looking for if you don't want write to memory map manually

commit bb2c4eb8 (2018-03-05): cleanup: remove old nrf51 codegen script

The script is no longer being used, and isn't particularly useful
anymore as the code it needs to generate has changed.

Fixes #726
closed 2018-03-05 by alevy