tock/tock #731: Don't generate asm listing in debug target

Grawp, 2018-01-21:
https://github.com/helena-project/tock/blob/27f45c54705c92ae9c7967634bdc14a48de6997f/boards/Makefile.common#L153

It is considerably slowing the debug builds and there is already **lst** target in the Makefile.

Moreover if you use `xargo` generating [unoptimized + debuginfo] sysroot (i.e. xargo compiled with _dev_ feature, the listing generation takes tens of minutes which renders the **debug** target unusable.

alevy, 2018-01-21:
I am personally in favor of this (I always just use objdump anyway), but this has been hotly* debated on the past.

*Our debates are never really very hot. I also personally never use the debug builds.

ppannuto, 2018-01-21:
A quick interlude independent of decision here, are you running on a Mac? It's known to be artifically slow and there's a fix for that here: https://github.com/helena-project/tock/blob/master/doc/Getting_Started.md#heads-up

brghena, 2018-01-21:
I'm also just fine with the `lst` target being removed from the debug build, especially if it's causing problems for you. Over time it's been added back in / removed somewhat randomly.

One question for you, is what you are using the debug build for? For example, I'm fine with `lst` targets being run in debug builds since I only build in debug mode when I need to actually debug a particular problem, which too often seems to involve reading through some assembly too. But we should make sure that the semantics of `make debug` match what users are doing with it, which is possibly not the same as me.

Grawp, 2018-01-22:
@brghena I use _Visual Studio Code_ (the opensource javascript thingy, not _Visual Studio_) with [native debug](https://marketplace.visualstudio.com/items?itemName=webfreak.debug) and [rls-vscode](https://github.com/rust-lang-nursery/rls-vscode) to visually step through the tock code in order to learn how tock works or to help me with porting effort.

I've thought that was the point of debug build i.e. same as in C/C++ development for MCU. You connect via SWD or JTAG, place breakpoints in your code, interrupt handlers, wherever and then step in/over/out, observe variables, registers content etc. in your IDE.

(I also use xargo with "dev" feature on, as I've written above, which generates debug build of sysroot so I can also step through *core* library. But it shouldn't be necessary. I haven't tried linking unoptimized rust code with optimized sysroot yet though.)

I certainly don't want `lst` target removed. I just want to suggest to separate generating listing from `debug` target i.e. make two targets `debug` and `lst-debug` or something like that.
It's not much of an issue for me to keep having it locally patched.

brghena, 2018-01-22:
Yeah, I'm fully in favor of this change then. The debug build should support JTAG without taking a ridiculous amount of time, and having a `lst-debug` target (or something like that) is easy enough to use.

bradjc, 2018-03-05:
Did this ever get added?

ppannuto, 2018-03-06:
https://github.com/helena-project/tock/blob/3333dbac72aa7c70731480d9d3514817e6081e3b/boards/Makefile.common#L158

Looks like no

commit 3d4d3c8a (2018-05-10): boards: debug: build lst seperately

This commit adds a make target for boards called `debug-lst` which
builds the debug version LST file.

It also disables LTO on debug builds since when it is set to true the
debug builds don't compile and throw this error:

```
make debug
   Compiling kernel v0.1.0 (file:///Users/bradjc/git/tock/kernel)
   Compiling imix v0.1.0 (file:///Users/bradjc/git/tock/boards/imix)
   Compiling cortexm4 v0.1.0 (file:///Users/bradjc/git/tock/arch/cortex-m4)
   Compiling capsules v0.1.0 (file:///Users/bradjc/git/tock/capsules)
   Compiling sam4l v0.1.0 (file:///Users/bra [... cut]
PR #928 (2018-05-10): boards: debug: build lst seperately
This commit adds a make target for boards called `debug-lst` which
builds the debug version LST file.

It also disables LTO on debug builds since when it is set to true the
debug builds don't compile and throw this error:

```
make debug
   Compiling kernel v0.1.0 (file:///Users/bradjc/git/tock/kernel)
   Compiling imix v0.1.0 (file:///Users/bradjc/git/tock/boards/imix)
   Compiling cortexm4 v0.1.0 (file:///Users/bradjc/git/tock/arch/cortex-m4)
   Compiling capsules v0.1.0 (file:///Users/bradjc/git/tock/capsules)
   Compiling sam4l v0.1.0 (file:///Users/brad [... cut]
commit 7a49b48d (2018-05-11): boards: debug: build lst seperately

This commit adds a make target for boards called `debug-lst` which
builds the debug version LST file.

It also disables LTO on debug builds since when it is set to true the
debug builds don't compile and throw this error:

```
make debug
   Compiling kernel v0.1.0 (file:///Users/bradjc/git/tock/kernel)
   Compiling imix v0.1.0 (file:///Users/bradjc/git/tock/boards/imix)
   Compiling cortexm4 v0.1.0 (file:///Users/bradjc/git/tock/arch/cortex-m4)
   Compiling capsules v0.1.0 (file:///Users/bradjc/git/tock/capsules)
   Compiling sam4l v0.1.0 (file:///Users/bra [... cut]
closed 2018-05-10 by alevy