tock/tock #734: [NRF52-DK] MPU bug

niklasad1, 2018-01-25:
Hi,

I wanted to test the MPU feature introduced in [#713](https://github.com/helena-project/tock/pull/713), where two apps in the following sequence were loaded:
1. ```blink```
2. ```MPU Walk Regions```

To be clear I expect the ```MPU Walk Regions``` to crash with a stack trace printed and the ```blink``` app to continue to execute but that's not happening!

Instead I see the following printout in ```tockloader```:
```bash
[TEST] MPU Walk Regions

  app_memory:            0x20002000-0x20004000
  app_grant:             0x20003ef4-0x20004000
  app_memory_accessible: 0x20002000-0x20003e00
  app_flash:             0x20000-0x22000

Walking flash
  ! Will overrun

0x20000 -> 0x23000, incr 0x100
       CPSR: 00000003
0x20000: 00000002
0x20100: 000000db
0x20200: 00000059
0x20300: 00000005
0x20400: 00000042
0x20500: 00000010
0x20600: 00000061
0x20700: 0000008c
0x20800: 00000070
0x20900: 00000065
0x20a00: 00000002
0x20b00: 0000001c
0x20c00: 00000004
0x20d00: 00000043
0x20e00: 0000004b
0x20f00: 0000002a
0x21000: 00000043
0x21100: 00000070
0x21200: 00000037
0x21300: 00000006
0x21400: 00000000
0x21500: 0000009b
0x21600: 00000000
0x21700: 00000032
0x21800: 00000002
0x21900: 0000007d
0x21a00: 00000000
0x21b00: 00000073
0x21c00: 00000050
0x21d00: 00000000
0x21e00: 00000000
0x21f00: 00000000
0x22000: 
```

Afterwards I also checked with gdb and the kernel is stuck in the [```hard_fault_handler```]( https://github.com/helena-project/tock/blob/master/chips/nrf52/src/crt1.rs#L32-#L34)

@alevy 
Can you confirm this on your HW also whether it's a bug or not?

alevy, 2018-01-25:
Yes, this is expected. That's the MPU catching the problem. There is no stack trace because the NRF52's `hard_fault_handler` doesn't panic (it just loops).

alevy, 2018-01-25:
Sorry, also the scheduler is currently not set up to continue other apps, it just panics. There is a setting you can use to allow other apps to resume, but again, the `hard_fault_handler` on the nrf52 still needs to be modified.

niklasad1, 2018-01-26:
Ok, rename the issue or close?

alevy, 2018-01-27:
Ah, good point... Rename (and I guess readd the bug label) or close and repoen a different one

bradjc, 2018-05-22:
I think this has been resolved.

closed 2018-05-22 by bradjc