tock/tock #737: Building C / C++ code into Tock project

Niondir, 2018-01-29:
Hey, not sure if this is the right place to ask.

I have a bunch of C and C++ code that is mostly free of any RTOS (and the parts that are could be migrated pretty easily). I'm struggling with using Tock because I would need a lot of the existing C code to work efficient on projects. It would help a lot if there is any tutorial about how to make a mixed Rust project, containing C and Rust code.

The `main()` and business logic would be in Rust, but a lot of drivers, board specific code and the HAL must be used from C and can be migrated step by step later.
We currently doing this with the migration from C to C++ where it is fairly easy, but Rust would be a nice next step.

**I see 2 options:**
a) Use the rust Build System to also build and link the C / C++ code 
b) use CMake (which I currently use for C and C++) and add support for Rust

Any hints on whats the "best practice" are very welcome. In case I get this running one day I would also prepare a tutorial for others, if there isn't already one.

P.S. the ROTS we are using in C ist "FreeRTOS" having a TOCK Wrapper or migration strategy would also be a great thing, but I guess that's not too urgent right now.

bradjc, 2018-01-29:
If you haven't seen these yet, it may be helpful to read through the overview and design documents:
- https://github.com/helena-project/tock/blob/master/doc/Overview.md
- https://github.com/helena-project/tock/blob/master/doc/Design.md

In Tock, we only support C/C++ code as processes running in userspace. Processes have to interact with the kernel through syscalls and a context switch, but we do have userspace interfaces for many common things. See https://github.com/helena-project/tock/tree/master/doc/syscalls for a reasonably complete list of interfaces that userpsace can use (conditional on hardware support, of course).

Our examples https://github.com/helena-project/tock/tree/master/userland/examples folder has many C apps (and one C++ app) that may be useful inspiration. As for porting from other systems, we currently use a Nordic BLE interface (https://github.com/helena-project/tock/tree/master/userland/libnrfserialization) that is all C code that runs in a process (e.g. https://github.com/helena-project/tock/blob/master/userland/examples/ble-uart/main.c). We have a custom kernel capsule (https://github.com/helena-project/tock/blob/master/capsules/src/nrf51822_serialization.rs) that directly supports the existing C library. That may be an approach that would work well for you: keep most of the code in c/c++ and write a small kernel-level capsule to support it.

Niondir, 2018-01-31:
I still don't fully understand how the build process works. As far as I can see, each process has it's own main function. How would you link multiple processes into a single binary? And I don't understand how the `ble-uart/main.c` and `nrf51822_serialization.rs` are connected.

And from the Architecture overview it looks like every process also has it's own heap. Is it be possible to share the heap across multiple processes?

brghena, 2018-01-31:
In Tock, each process is a standalone unit, with its own heap, stack, and data segment. Processes interact with the kernel through system calls, but are prevented (by the MPU) from accessing memory that doesn't belong to them.

For example, ble-uart is a process in Tock and has a main function and various code that runs. Some of that code uses other C libraries (like the Nordic BLE libraries). At the bottom of the library though, to actually effect a change in hardware the process must perform a system call. In this case, the call is linked up to the nrf51822_serialization.rs capsule, which handles actually modifying hardware state (by sending messages over a UART).

Multiple processes cannot share memory, but can send messages between each other with Tock's inter-process communication (IPC) mechanism (https://github.com/helena-project/tock/blob/7299d6c339d62c0fb6bfc0b823ef17065a87a2bf/doc/Userland.md#inter-process-communication).

ppannuto, 2018-01-31:
> How would you link multiple processes into a single binary? 

In addition to Branden's great response I'd add that this is an explicitly against the Tock ethos. If you want to have your process do multiple things, that's totally fine and you can write code to do so, but processes are independent, just like on a desktop operating system.

brghena, 2018-01-31:
For use cases where users have a lot of existing C/C++ code, the goal is usually to push that functionality into an application, with as few connections as possible to the kernel. Logic and control can be performed in the application, but processes are not capable of directly accessing memory-mapped I/O, so the kernel needs to do that part. This means at least the most bottom layers of code which access the hardware need to be ported to Rust.

While you could imagine a build system that combines C/C++ code with Rust code for the kernel and low-level accesses, we do not do this in Tock. The problem is really guaranteeing memory safety. Rust has several capabilities to analyze code and ensure certain guarantees, but it can only do so if it's in charge of everything. Introducing C code into the mix ends up invalidating a lot of the guarantees.

For people working on existing supported platforms, this shouldn't be an issue. We've already written a lot of drivers in Rust for you, and you can write your application in C (or C++, or Lua, or Rust, or a lot of other choices if you're willing to get the toolchains for them set up and have enough memory) and not have to think about it.

For supporting new platforms though, developers have to get their hands dirty and write at least a little bit of rust code (a driver for each peripheral they want to use). This is certainly a bummer for people who have already written drivers in a language they're more comfortable with (and this certainly included me a couple years ago), but we've found that the benefits of having a secure kernel and the error-checking capabilities of Rust have outweighed the costs of porting code.

Niondir, 2018-02-02:
These answers already help a lot, thanks. What I take from this (please correct me if I'm wrong):

* It does not make sense to run more than one process on a single constraint device, but the process can use e.g. the "Thread" capsule, to do stuff in parallel.
* A process should not (or with MPU can not) access any memory mapped I/O but must use syscalls to capsules. (we don't have an MPU)
* Since all capsules run in a single loop, they must not block. So all hardware access must be implemented without blocking.

**On example:**
We have an XMAC based duty cycle protocol which uses an SX1272 radio tranceiver.
In Tock I would have to migrate all the hardware access to the SX1272 into a Rust Capsule using another capsule for the SPI based communication. Where would I implement the time constraint duty cycle protocol. In FreeRTOS we use a high priority Task that is preemptively scheduled.

> While you could imagine a build system that combines C/C++ code with Rust code for the kernel and low-level accesses, we do not do this in Tock.

Does this means I end up with 2 binaries, the Tock OS and my "process" and rely on a Tock bootloader to flash my process?

alevy, 2018-02-02:
> It does not make sense to run more than one process on a single constraint device, but the process can use e.g. the "Thread" capsule, to do stuff in parallel.

There can definitely be multiple processes, which is generally the way to achieve multiple independent tasks in userspace. I'm not sure what you mean the "Thread" capsule.

> A process should not (or with MPU can not) access any memory mapped I/O but must use syscalls to capsules. (we don't have an MPU)

Correct

> Since all capsules run in a single loop, they must not block. So all hardware access must be implemented without blocking.

Correct. In some edge cases it might technically be OK to block for a short time (for example, on the SAM4L the GPIO controller has a busy flag that we busy wait on because it never takes more than a couple cycles), but generally the kernel APIs should be non-blocking and blocking implementations are only the very rare and well justified exceptions.

> Does this means I end up with 2 binaries, the Tock OS and my "process" and rely on a Tock bootloader to flash my process?

I think there are actually two separate questions hidden in this questions: (1) which kinds of programs run on a Tock platform? (2) how do you write those programs onto the MCU flash?

You definitely end up, logically, with at least two binaries: one for the kernel, and one or more for processes. In general, on the platforms we use in this repo, there is a bootloader (called the "tock bootloader") that helps program each of those types of binaries.

However, it *is* technically possible to take the kernel ELFs and process ELFs, and combine them into a single binary that you program in one shot onto the MCU... We've done that at various points, and there may even be some script in some Makefile with remenants of that.

> One example: ...

As to your XMAC example: I think you're right, so I'll just reiterate to make sure I understood you. In Tock, there would be a number of capsules: SPI, virtual SPI (maybe, if the SPI controller is used by several drivers), the SX1272 capsule, and the XMAC capsules (as well as maybe capsules for other layers of the network protocol). These would all run in the kernel with a higher priority than any process, and would serve the role of the high priority FreeRTOS thread.

You might be interested to look at the XMAC driver that was just merged. The stack we currently have in master for the `imix` platform uses the Atmel RF233 radio (`capsules/src/rf233.rs`), also attached with SPI. Everything _should_ be separated in such a way that for your use cases, you ought to only need to implement a capsule for the SX1272 that implements the traits in `kernel::hil::radio`. That should plug in to the rest of the stack.

alevy, 2018-02-02:
I should mention, it _is_ possible to use SPI directly from userspace if you use the `capsules/src/spi.rs` system call driver (itself a capsule). This would let you use a C/C++ implementation of everything from the SX1272 driver and up. That's just not _cannonical_ Tock.

This was how the rf233 driver started out before migrating it to the kernel, and how the NRF51 is currently driven on the hail board (using nordic's serilization library in C in a process)

Niondir, 2018-02-03:
Cool I now get a picture of everything :)

Multiple processes are possible but due to very limited ram and text section they would waste a lot of memory and flash. 
@Threads: I just saw that there is a thread capsule in the overview picture and assumed that it can be used by a single processes to run stuff in parallel (preemptively? with priorities?).

I got some ideas how slowly migrating from a pure unsafe C/C++ codebase could work without having all the safety guarantees in place in the beginning. 
It would just need some concepts in usersapce supported by tock like: Threads (maybe with priorities), delays, queues / notifications / semaphores  to communicate between Threads, software timers.
With that features I could replace all the parts where we currently use FreeRTOS and then start to migrate all the hardware access into using Tock.

Still not sure if it's worth it. But I love the idea of using rust on embedded and starting from scratch is not an option.