tock/tock #756: rustdoc deprecations

ppannuto, 2018-02-26:
Looks like rustdoc has changed a bit and things are going to break soon:

```
 Documenting nrf52dk v0.1.0 (file:///Volumes/code/helena-project/tock/boards/nrf52dk)
WARNING: the 'no-defaults' flag is considered deprecated
WARNING: please see https://github.com/rust-lang/rust/issues/44136
WARNING: the 'passes' flag is considered deprecated
WARNING: please see https://github.com/rust-lang/rust/issues/44136
WARNING: (you may want to use --document-private-items)
```

There's also several hundred warnings emitted by rustdoc when running `make doc` (after #755 fixes), largely of the form like:

```
WARNING: rendering difference in `Query and control pin values and states.`
   --> /Volumes/code/helena-project/tock/capsules/src/gpio.rs:175:4
    /html[0]/body[1]/p[2] Unexpected element `code`: found: `<code>pin_config</code>`
WARNING: rendering difference in `Query and control pin values and states.`
   --> /Volumes/code/helena-project/tock/capsules/src/gpio.rs:175:4
    /html[0]/body[1]/p[2] One element is missing: expected: `: An internal resistor setting.`
```

ppannuto, 2018-02-27:
#755 fixed most of these, but there are still some lingering warnings for nrf51/2 and ble.

To see them, run `tools/build-all-docs.sh`. #755 should give guidance for how to fix the remaining fellows.

niklasad1, 2018-02-28:
Thanks @ppannuto haha I will take of that soon :see_no_evil:

closed 2018-03-03 by ppannuto