tock/tock #763: Question, is that possible add ARM7  support in tock?

lygstate, 2018-02-28:
For some IoT devices that broadly used such as 
https://labs.mediatek.com/en/chipset/MT2503
&
https://labs.mediatek.com/en/chipset/MT2502
They are still using ARM7 architecture with big memory(4MBytes)
Is that possible to getting tock support for those chipset.

alevy, 2018-03-05:
In principle yes, with a one caveat: we would need to port the process isolation to use the virtual memory on those architectures. There isn't nothing architecturally preventing us from doing that, it's just some amount of engineering work someone's gotta do.

If someone is interested in pursuing that, I think it would make sense and I believe we would all be supportive of it.

lygstate, 2018-03-06:
yeap I have interest to do that

lygstate, 2018-03-06:
I have a question， do we need to rebuild newlib for arm7， cause the instruction set for arm7 are arm5vt

ppannuto, 2018-03-06:
Not to get started, no. M7 should be a superset of M4. Eventually it would
be efficient to rebuild, but in the short term the m4 build is fine

brghena, 2018-03-06:
@ppannuto I think you misunderstood. The question is about ARM7, not a Cortex-M7. This is the old class of ARM processors.