tock/tock #778: Enable rust doc verification in CI

niklasad1, 2018-03-06:
I think we should enable ```make alldocs``` in Travis to catch Rust doc errors instead of manually inspect each PR. 

/cc @ppannuto @bradjc

bradjc, 2018-03-06:
We could try it, and see how long the time penalty for adding it is.

It would help catch errors when changing rust versions and the docs stop building. It will also print warnings in kind of odd cases, some of which are actually worth fixing. It doesn't, unfortunately, check that the docs "look ok". That would still have to be done manually.

ppannuto, 2018-03-06:
Aha! We are! (As of recently :) )

If there’s a doc building error, the netlify build will fail. We can
upgrade that to a fail on warnings too if we like.

Currently netlify isn’t marked as a required check, but we probably should
upgrade it to be required

closed 2018-03-10 by niklasad1