tock/tock #798: [Netlify/docs-tockosorg] Configuration error

niklasad1, 2018-03-10:
The Netlify configuration seems to be broken:

``` bash
1:15:46 AM: error: intermittent IO error while iterating directory `/opt/buildhome/.rustup/toolchains/nightly-2018-01-05-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/bin`
1:15:46 AM: caused by: IO error for operation on /opt/buildhome/.rustup/toolchains/nightly-2018-01-05-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/bin: No such file or directory (os error 2)
1:15:46 AM: caused by: No such file or directory (os error 2)
1:15:46 AM: note: run with `RUST_BACKTRACE=1` for a backtrace
1:15:46 AM: make: *** [doc] Error 1
```

ppannuto, 2018-03-10:
Ugh.., looks like something changed upstream. I can confirm this breaks locally if you update to xargo 0.3.11, which was just released. Looking into a fix now

ppannuto, 2018-03-11:
Fixed by #799

closed 2018-03-11 by ppannuto