tock/tock #817: Bad app header can crash kernel

bradjc, 2018-03-17:
Trying out https://github.com/tock/libtock-rs/pull/18 I can get panics in the kernel:

```
Kernel panic at /Users/bradjc/git/tock/arch/cortex-m4/src/mpu.rs:164:
	"attempt to calculate the remainder with a divisor of zero"
	Kernel version release-1.0-2018-02-161-ga11bbe72

---| Fault Status |---
No faults detected.

---| App Status |---
App: blink   -   [Running]
 Events Queued: 0   Syscall Count: 0   Last Syscall: None

 ╔═══════════╤══════════════════════════════════════════╗
 ║  Address  │ Region Name    Used | Allocated (bytes)  ║
 ╚0x20004000═╪══════════════════════════════════════════╝
             │ ▼ Grant         280 |    280
  0x20003EE8 ┼───────────────────────────────────────────
             │ Unused
  0x20004040 ┼───────────────────────────────────────────
             │ ▲ Heap       536887360 | 536887016 EXCEEDED!     S
  0x00000000 ┼─────────────────────────────────────────── R
             │ Data              0 |      0               A
  0x00000000 ┼─────────────────────────────────────────── M
             │ ▼ Stack      3758079968 | 3758080000
  0x20004020 ┼───────────────────────────────────────────
             │ Unused
  0x20004000 ┴───────────────────────────────────────────
             .....
  0x00031000 ┬─────────────────────────────────────────── F
             │ App Flash      4052                        L
  0x0003002C ┼─────────────────────────────────────────── A
             │ Protected        44                        S
  0x00030000 ┴─────────────────────────────────────────── H

  R0 : 0x0003002C    R6 : 0x00000000
  R1 : 0x20004000    R7 : 0x00000000
  R2 : 0x00000000    R8 : 0x00000000
  R3 : 0x20004040    R10: 0x00000000
  R4 : 0x00000000    R11: 0x00000000
  R5 : 0x00000000    R12: 0x00000000
  R9 : 0x00000000 (Static Base Register)
  SP : 0x20004020 (Process Stack Pointer)
  LR : 0x000309EF
  PC : 0x000309EF
 YPC : 0x000309EF

 APSR: N 0 Z 0 C 0 V 0 Q 0
       GE 0 0 0 0
 IPSR: Exception Type - Thread Mode
 EPSR: ICI.IT 0x00
       ThumbBit true
 To debug, run `make debug RAM_START=0x20004000 FLASH_INIT=0x309ef`
 in the app's folder and open the .lst file.
```

```
$ tockloader info
No device name specified. Using default "tock"
Using "/dev/cu.usbserial-1410 - Hail IoT Module - TockOS"

     [Info]: Paused an active tockloader listen in another session
tockloader version: 0.10.0-dev
Showing all properties of the board...
Apps:
[App 0]
  Name:                  blink
  Enabled:               True
  Sticky:                False
  Total Size in Flash:   4096 bytes
  Address in Flash:      0x30000
    package_name          : blink
    version               :        2
    checksum              :            0x6e4c75cb
    flags                 :        1          0x1
      enabled             : Yes
      sticky              : No
    header_size           :       44         0x2c
    init_fn_offset        :     2499        0x9c3
    minimum_ram_size      :        0          0x0
    protected_size        :        0          0x0
    total_size            :     4096       0x1000

Attributes:
00:    board = hail
01:     arch = cortex-m4
02: jldevice = ATSAM4LC8C
03:       id = c098e513000a
04:
05:
06:
07:
08:
09:
10:
11:
12:
13:
14:
15:

Bootloader version: 0.6.0
Finished in 2.269 seconds
     [Info]: Resumed other tockloader listen session
```

```
$ hexdump -C target/cortex-m4.bin
00000000  02 00 2c 00 00 10 00 00  01 00 00 00 cb 75 4c 6e  |..,..........uLn|
00000010  01 00 0c 00 c3 09 00 00  00 00 00 00 00 00 00 00  |................|
00000020  03 00 05 00 62 6c 69 6e  6b 00 00 00 80 b5 02 20  |....blink...... |
00000030  00 21 00 22 00 23 02 df  00 28 09 dd 02 20 03 21  |.!.".#...(... .!|
00000040  00 22 00 23 02 df 4f f4  fa 70 00 f0 b7 f8 f5 e7  |.".#..O..p......|
00000050  4f f2 64 10 cf f6 ff 70  78 44 00 f0 d4 fb fe de  |O.d....pxD......|
00000060  2d e9 f0 41 84 b0 04 46  65 68 75 b1 6f 00 b7 f1  |-..A...Fehu.o...|
00000070  ff 3f 42 dd d4 f8 00 80  af 42 14 d1 46 46 2f 46  |.?B......B..FF/F|
00000080  c4 e9 00 67 04 b0 bd e8  f0 81 49 46 68 46 04 22  |...g......IFhF."|
00000090  01 23 00 f0 d7 fa 00 98  01 28 1c bf 01 9e 00 2e  |.#.......(......|
```

Woyten, 2018-03-17:
Are you using `rustup override`? If yes, you can try unsetting the override and let the toolchain file pick the correct toolchain. `nightly-2018-03-06` works fine on our NRF52DK board (with or without LLD).

bradjc, 2018-03-18:
I deleted the override.

```
$ pwd
/Users/bradjc/git/libtock-rs
$ rustc --version
rustc 1.26.0-nightly (c9334404f 2018-03-05)
```

I have to do something like this to get it to work:

```diff
diff --git a/layout.ld b/layout.ld
index 4df68f9..fe9e0ef 100644
--- a/layout.ld
+++ b/layout.ld
@@ -4,9 +4,7 @@
  * is not known. Therefore, this script over provisions space on some platforms.
  */

-STACK_SIZE = 2048;
-APP_HEAP_SIZE = 1024;
-KERNEL_HEAP_SIZE = 1024;
+

 ENTRY(_start)

@@ -108,17 +107,17 @@ SECTIONS {
  */
     .stack :
     {
-        . += STACK_SIZE;
+        . += 2048;
     } > SRAM

     .app_heap :
     {
-        . += APP_HEAP_SIZE;
+        . += 1024;
     } > SRAM

     .kernel_heap :
     {
-        . += KERNEL_HEAP_SIZE;
+        . += 1024;
     } > SRAM

     _sram_end = .;
diff --git a/run_example.sh b/run_example.sh
index c6e7d0e..5da5a3b 100755
--- a/run_example.sh
+++ b/run_example.sh
@@ -8,7 +8,10 @@ tab_file_name=metadata.toml
 elf_file_name=cortex-m4.elf
 bin_file_name=cortex-m4.bin

-RUST_TARGET_PATH=$(pwd) cargo run --manifest-path xargo/Cargo.toml -- build --release --target=thumbv7em-tock-eabi --example "$1"
+RUSTFLAGS_FOR_XARGO_LINKING="-C link-arg=-Tlayout.ld"
+
+RUST_TARGET_PATH=$(pwd) cargo run --manifest-path xargo/Cargo.toml -- build --release --target=thumbv7em-tock-eabi --verbose --example "$1"
+RUST_TARGET_PATH=$(pwd) xargo build --release --target=thumbv7em-tock-eabi --verbose --example "$1"
 cp target/thumbv7em-tock-eabi/release/examples/"$1" "target/$elf_file_name"
 cargo run --manifest-path tock/userland/tools/elf2tbf/Cargo.toml -- -n "$1" -o "target/$bin_file_name" "target/$elf_file_name"
```

Woyten, 2018-03-18:
I will respond on tock/libtock-rs/pull/18.

closed 2018-04-16 by bradjc