tock/tock #818: Userland: `make debug` creates LST file at wrong address

brghena, 2018-03-18:
A puzzle that Josh and I ran into tonight and finally managed to distill. When you build LST files with `make debug RAM_START=... FLASH_INIT=...`, they sometimes work correctly, and are sometimes off by 4 bytes.

For example, `blink` has a correct LST file:
```
$ cd userland/examples/blink/
$ make debug RAM_START=0x00000000 FLASH_INIT=0x80000000
$ head -n  72 build/cortex-m4/cortex-m4.userland_debug.lst | tail -n 30

Disassembly of section .text:

7fffffdc <_text>:
7fffffdc:	039c      	lsls	r4, r3, #14
7fffffde:	0000      	movs	r0, r0
7fffffe0:	0000      	movs	r0, r0
7fffffe2:	0000      	movs	r0, r0
7fffffe4:	0024      	movs	r4, r4
7fffffe6:	0000      	movs	r0, r0
7fffffe8:	03c0      	lsls	r0, r0, #15
7fffffea:	0000      	movs	r0, r0
7fffffec:	0024      	movs	r4, r4
7fffffee:	0000      	movs	r0, r0
7ffffff0:	0000      	movs	r0, r0
7ffffff2:	0000      	movs	r0, r0
7ffffff4:	0024      	movs	r4, r4
7ffffff6:	0000      	movs	r0, r0
7ffffff8:	014c      	lsls	r4, r1, #5
7ffffffa:	0000      	movs	r0, r0
7ffffffc:	03c0      	lsls	r0, r0, #15
	...

80000000 <_start>:
__attribute__ ((weak))
__attribute__ ((noreturn))
void _start(void* text_start,
            void* mem_start,
            void* memory_len __attribute__((unused)),
            void* app_heap_break __attribute__((unused))) {
80000000:	b508      	push	{r3, lr}
```

But `find_north` has a bad LST file:
```
$ cd userland/examples/find_north
$ make debug RAM_START=0x00000000 FLASH_INIT=0x80000000
$ head -n  72 build/cortex-m4/cortex-m4.userland_debug.lst | tail -n 30

Disassembly of section .text:

7fffffe0 <_text>:
7fffffe0:	32d8      	adds	r2, #216	; 0xd8
7fffffe2:	0000      	movs	r0, r0
7fffffe4:	0000      	movs	r0, r0
7fffffe6:	0000      	movs	r0, r0
7fffffe8:	008c      	lsls	r4, r1, #2
7fffffea:	0000      	movs	r0, r0
7fffffec:	3364      	adds	r3, #100	; 0x64
7fffffee:	0000      	movs	r0, r0
7ffffff0:	008c      	lsls	r4, r1, #2
7ffffff2:	0000      	movs	r0, r0
7ffffff4:	0069      	lsls	r1, r5, #1
7ffffff6:	0000      	movs	r0, r0
7ffffff8:	00f8      	lsls	r0, r7, #3
7ffffffa:	0000      	movs	r0, r0
7ffffffc:	0170      	lsls	r0, r6, #5
7ffffffe:	0000      	movs	r0, r0
80000000:	33cd      	adds	r3, #205	; 0xcd
	...

80000004 <_start>:
__attribute__ ((weak))
__attribute__ ((noreturn))
void _start(void* text_start,
            void* mem_start,
            void* memory_len __attribute__((unused)),
            void* app_heap_break __attribute__((unused))) {
80000004:	b508      	push	{r3, lr}
```

A diff of the `cortex-m4.userland_debug.ld` files that are generated shows them to be identical. So why does linking different code with the same LD file end up with `_start` at a different location??

alevy, 2018-03-18:
Somewhere in the back of my mind I think I might know the answer to this. Assigned to myself so i remember to look into this.

bradjc, 2018-03-18:
Ugh why does the compiler sometimes put `_start` at one place and then other times (namely when we run `make debug`) put `_start` four bytes later? Our math doesn't expect this and hence the debug version of the LST is in the wrong spot.

bradjc, 2018-05-22:
Has this just fixed itself? When I run on find_north:

```
$ cd userland/examples/find_north
$ make debug RAM_START=0x00000000 FLASH_INIT=0x80000000
$ head -n  1400 build/cortex-m4/cortex-m4.userland_debug.lst | tail -n 30

Disassembly of section .wfr.app_state:

7fffffff <.wfr.app_state>:
	...

Disassembly of section .text:

80000000 <_start>:
__attribute__ ((weak))
__attribute__ ((noreturn))
void _start(void* app_start,
            void* mem_start,
            void* memory_len __attribute__((unused)),
            void* app_heap_break __attribute__((unused))) {
80000000:	4604      	mov	r4, r0
  // Allocate stack and data. `brk` to stack_size + got_size + data_size +
  // bss_size from start of memory. Also make sure that the stack starts on an
  // 8 byte boundary per section 5.2.1.2 here:
  // http://infocenter.arm.com/help/topic/com.arm.doc.ihi0042f/IHI0042F_aapcs.pdf
  struct hdr* myhdr = (struct hdr*)app_start;
  uint32_t stacktop = (((uint32_t)mem_start + myhdr->stack_size + 7) & 0xfffffff8);
80000002:	1dcd      	adds	r5, r1, #7
80000004:	6a41      	ldr	r1, [r0, #36]	; 0x24

  // fix up GOT
  volatile uint32_t* got_start     = (uint32_t*)(myhdr->got_start + stacktop);
  volatile uint32_t* got_sym_start = (uint32_t*)(myhdr->got_sym_start + (uint32_t)app_start);
80000006:	6822      	ldr	r2, [r4, #0]
  volatile uint32_t* got_start     = (uint32_t*)(myhdr->got_start + stacktop);
```

It looks ok. I also tried on other apps and also am seeing the correct address.

ppannuto, 2018-05-22:
> they sometimes work correctly, and are sometimes off by 4 bytes.

It could be that `find_north` used to be bad and then other changes put it on the right side of history? Did we ever track down the root cause of this? @alevy any recollection of what was in the back of your mind?

bradjc, 2018-06-06:
Tried with find_north, hail, ble-env-sense, all seem fine. I've made a note that we should make this a test on the `libtock-c` repo in case this ever regresses.

closed 2018-06-06 by bradjc