tock/tock #820: NRF51: hard fault on application start

fhars, 2018-03-18:
As far as I can tell, tock on an NRF51 ends in the hard fault handler shortly after it starts to run any application starting with commit 548726fb22ef15b2c8d498b01200c7e446c212e5 "WIP - blink working"

I made a minmal board definition with just the TemperatureSensor, Console, AlarmDriver and SimpleRng capsules, and both the sensors application and the blink application hard fault, sensors in the first switch_to_user call, blink in the second. Without an application, the kernel seems to idle along just fine.

Here is the stack trace (for whatever gdb stack traces of rust code are worth) for sensors:
```
#0  nrf51::crt1::hard_fault_handler () at /home/hars/tock/chips/nrf51/src/crt1.rs:38
#1  <signal handler called>
#2  0x0000347a in cortexm0::switch_to_user (user_stack=0x200017b0 <microbit::APP_MEMORY+32> "\320\027\000", process_regs=0x200015c8 <microbit::PROCESSES+56>)
    at /home/hars/tock/arch/cortex-m0/src/lib.rs:109
#3  kernel::process::Process::switch_to (self=<optimized out>) at /home/hars/tock/kernel/src/process.rs:1166
#4  kernel::sched::do_process<microbit::Platform,nrf51::chip::NRF51> (process=<optimized out>, platform=<optimized out>, chip=<optimized out>, appid=..., ipc=<optimized out>)
    at /home/hars/tock/kernel/src/sched.rs:40
#5  kernel::main::{{closure}}<microbit::Platform,nrf51::chip::NRF51> (process=<optimized out>) at /home/hars/tock/kernel/src/lib.rs:57
#6  core::option::Option<&mut kernel::process::Process>::map<&mut kernel::process::Process,(),closure> (self=..., f=...)
    at /home/hars/.rustup/toolchains/nightly-2017-11-19-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/src/libcore/option.rs:398
#7  kernel::main<microbit::Platform,nrf51::chip::NRF51> (processes=<optimized out>, platform=<optimized out>, chip=<optimized out>, ipc=<optimized out>) at /home/hars/tock/kernel/src/lib.rs:56
#8  microbit::reset_handler () at src/main.rs:203
#9  0x000026c4 in microbit::io::rust_begin_unwind (_args=..., _file=..., _line=<optimized out>) at src/io.rs:92
#10 0x00002000 in microbit::io::rust_begin_unwind (_args=..., _file=..., _line=<optimized out>) at src/io.rs:91
#11 0x200018a8 in microbit::APP_MEMORY ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

The last steps in the debugger:
```
(gdb)                                              
0x00020286 in ?? ()                                
(gdb)                                              
0x00020b78 in ?? ()                                
(gdb)                                              
0x00020b7a in ?? ()                                
(gdb)                                              
0x00020b7c in ?? ()                                
(gdb)                                              
0x00020b7e in ?? ()                                
(gdb)                                              
0x00020b80 in ?? ()                                
(gdb)                                              
0x00020b82 in ?? ()                                
(gdb)                                              
0x00020b84 in ?? () 
(gdb) info registers                               
r0             0x200018a8       536877224          
r1             0x200018a8       536877224          
r2             0x200017d0       536877008          
r3             0x74696e49       1953066569         
r4             0x200017d0       536877008          
r5             0x0      0                          
r6             0x0      0                          
r7             0x0      0                          
r8             0x0      0                          
r9             0x200017d0       536877008          
r10            0x0      0                          
r11            0x0      0                          
r12            0x0      0
sp             0x200017b0       0x200017b0 <microbit::APP_MEMORY+32>
lr             0x2028b  131723
pc             0x20b84  0x20b84
xpsr           0x1000000        16777216
msp            0x20001208       536875528
psp            0x200017b0       536876976
primask        0x0      0
control        0x2      2
(gdb) disassemble 0x20b78,0x20b88                  
Dump of assembler code from 0x20b78 to 0x20b88:    
   0x00020b78:  mov     r3, r9                     
   0x00020b7a:  mov     r2, r9                     
   0x00020b7c:  push    {r3, lr}                   
   0x00020b7e:  ldr     r3, [pc, #16]   ; (0x20b90)
   0x00020b80:  movs    r1, r0                     
   0x00020b82:  ldr     r3, [r2, r3]               
=> 0x00020b84:  ldr     r0, [r3, #0]               
   0x00020b86:  bl      0x20a9c                    
End of assembler dump.                             
(gdb) si
nrf51::crt1::hard_fault_handler () at /home/hars/tock/chips/nrf51/src/crt1.rs:38
38          'loop0: loop {}
```

alevy, 2018-03-18:
Were you sure to re-compile the process binaries completely from scratch (`make clean && make`)? The start ABI interface and process loading changed a month or two ago, and this problem sounds like it _could_ be due to running an old binary on a new kernel or vice-versa.

fhars, 2018-03-18:
I just tried again and made sure that yes, it happens after make clean; make in the app directory.

fhars, 2018-03-18:
Or is it no longer sufficient to just put the app at the correct location in flash?

alevy, 2018-03-18:
@fhars, that's enough. This doesn't _sound_ like you're doing anything wrong, though it's supposed to work :) I can try on my own nrf51 in a few hours.

alevy, 2018-03-25:
@hars can you post the actual binary you're using. This doesn't seem to correspond to the binary I'm getting when I compile the blink app. I'm expecting `r3` above to correspond to the `rel_data_offset` in the process's header, but I have a totally different value (`0x4c0` instead of `0x381`) when I compile blink.

fhars, 2018-03-28:
Something is trying to drive me mad, today I can no longer reproduce the problem with blink, but sensors does crash at the same point with a different odd number in r3:
```
(gdb) info registers                               
r0             0x22311  140049                     
r1             0x0      0                          
r2             0x200027b8       536881080          
r3             0x22e5   8933                       
r4             0x2002c  131116                     
r5             0x20002470       536880240          
r6             0x22198  139672                     
r7             0x0      0                          
r8             0x0      0                          
r9             0x20002470       536880240          
r10            0x0      0                          
r11            0x0      0                          
r12            0x0      0                          
sp             0x20002470       0x20002470 <microbit::APP_MEMORY+2048>                                
lr             0x200af  131247                     
pc             0x200b4  0x200b4 <_start+144>       
xpsr           0x1000000        16777216           
msp            0x20001880       536877184          
psp            0x20002470       536880240      0000120 b5f8 2380 011b 6942 18cd 6883 0004 189e
    
primask        0x0      0                          
control        0x2      2                          
```

I cannot read TBF headers (is there a tbfdump utility somewhere?), but I can find that number in the file:
```
$ od -t x2 build/cortex-m0/cortex-m0.bin | head -6
0000000 0002 002c 4000 0000 0001 0000 407d 733a
0000020 0001 000c 0025 0000 0000 0000 1745 0000
0000040 0003 0007 6573 736e 726f 0073 216c 0000
0000060 0000 0000 0110 0000 227c 0000 0110 0000
0000100 0069 0000 017c 0000 01cc 0000 22e5 0000
                                      ****
```

I attach the generated bin file
[cortex-m0.bin.gz](https://github.com/helena-project/tock/files/1857263/cortex-m0.bin.gz)

ppannuto, 2018-03-28:
Not tuned into the larger issues here, but if you `make V=1` when building
your app, or pass the `--verbose` flag to elf2tbf if you run it manually,
it'll print the TBF headers [an immediate solution in lieu of a tbfdump
utility, which does not currently exist]

fhars, 2018-03-30:
I just realized that the odd number in r3 is the sum of the .text and .data sizes, with .data being the odd one in the sum.

fhars, 2018-03-30:
Adding
```
        . = ALIGN(4); /* Make sure we're word-aligned here */
```
at the end of every section in `userland_generic.ld` makes the code run.

What is the trust model, should a broken application binary be able to crash the kernel?

alevy, 2018-03-30:
>  What is the trust model, should a broken application binary be able to crash the kernel?

On the NRF51, where there isn't an ARM MPU, yes. That's one of the reasons we have not been putting as much effort into that chip and rather focusing on the NRF52 (which does have an ARM MPU).

In general, no, and it won't (well technically the default policy of the kernel when an app crashes is to panic with a debug report, but that's just a policy and just killing the process or restarting it and letting the rest of the processes keep going is another option).

closed 2018-05-05 by bradjc