tock/tock #834: RFC: The Future of Tock Tools

bradjc, 2018-03-21:
This RFC is to discuss how we want the tools that Tock uses for building and flashing binaries to evolve as Tock progresses.

For the Tock ecosystem we currently have several Tock-specific tools:

- tockloader (python)
- elf2tbf (rust)
- create_tab (python)
- various makefiles (Make)

Some issues:

- elf2tbf has sort of evolved over time, and isn't particularly well architected due to the changes to app formats and what not.
- elf2tbf is sort of coupled with the userland folder in the main tock repo, yet other projects like libtock-rs need to use it.
- create_tab and elf2tbf are two separate tools.
- A lot of the makefile infrastructure for building c apps may need to be duplicated (in some form) for libtock-rs.
- How much should be rolled into tockloader? Assuming we want to keep the tooling in rust, and it's unlikely tockloader will be ported anytime soon, is it ok for generating TABs and using TABs to be two different tools?
- Should we move elf2tbf to its own repo? Put it on crates.io?

### One Idea

Have two tools, one for building TABs and one for using them. Try to push as much into the TAB building tool as possible to minimize what has to be copied for each app runtime/language. Put that tool in its own repo. Add to that tool the ability to publish TABs to some repo, and add to tockloader the ability to install apps from that repo.

ppannuto, 2018-03-22:
Regarding makefiles and duplication in libtock, I think/hope the long-term goal would be for that to be driven entirely by Cargo. I believe you should be able to use multiple `[[bin]]` entries in Cargo.toml to ask cargo to build for multiple architectures (eventually).

That said, we should draw a line of what the 'local userland build tool' is responsible for creating, most likely a .bin or .elf file for each architecture Tock supports

ppannuto, 2018-03-22:
As it is, `create_tab` is a [pretty minimal piece of software](https://github.com/helena-project/tock/blob/master/userland/tools/tab/create_tab.py).

Following the argument that the local build environment should like end at an elf, perhaps the `elf2tbf` tool should really be an `elfs2tab` tool. Or probably better, one `tock-tools` crate that provides `elf2tbf` and `tbfs2tab`. `create_tab` is small and simple enough that it's probably worth it to just port to rust

ppannuto, 2018-03-22:
I'm not sure what the vision is for the 'publish TAB to repo and load from repo', are you picturing something pypi-like? Or is this more of a local folder on machine is a "repo" type idea? It feels like you're trying to bridge the discovery gap between where one tool puts TABs and another finds them.

Right now we're doing that with directory hierarchies, and it seems to be working reasonably well. What's the motivation for changing?

alevy, 2018-03-24:
I'm in favor of having a single tool to build tabs: i.e. combine elf2tbf and create_tab.py. In general, I think it's perfectly fine for generating using tabs to be separate.

alevy, 2018-03-24:
One benefit of keeping generation and loading tabs separate is that other systems _could_ be ported to run Tock processes, but may very well have totally incompatible binary loading tools/mechanisms.

bradjc, 2018-06-20:
`elf2tbf` has been expanded to also create TABs, and is now called `elf2tab`. It is in its own repo and on crates.io, and is used directly by both libtock-c and libtock-rs.

Tockloader 1.0 has been released, and is gaining support for programming more boards.

closed 2018-06-20 by bradjc