tock/tock #864: Kernel Panic on Apps (or some segment of them) too large

adkinsjd, 2018-04-02:
I am currently having an issue where if I add code to my app, early in the app I get this kernel panic:

```
Kernel panic at /home/adkins/Research/signpost-software/signpost/kernel/tock/chips/sam4l/src/lib.rs:195:
	"Kernel HardFault.
	Kernel version v0.2-532-g3774e95
	r0  0x685c200e
	r1  0x961b287
	r2  0x598320a0
	r3  0xf1420a81
	r12 0x7b3bbcbe
	lr  0x17af80fe
	pc  0x9c6a7d3e
	prs 0xdca6ff97 [ N 1 Z 1 C 0 V 1 Q 1 GE 0110 ; ICI.IT 191 T false ; Exc 407-(Unknown! Illegal value?) ]
	sp  0x200051b0
	top of stack     0x200056b8
	bottom of stack  0x200036b8
	SHCSR 0x0
	CFSR  0x0
	HSFR  0x40000000
	Instruction Access Violation:       false
	Data Access Violation:              false
	Memory Management Unstacking Fault: false
	Memory Management Stacking Fault:   false
	Memory Management Lazy FP Fault:    false
	Instruction Bus Error:              false
	Precise Data Bus Error:             false
	Imprecise Data Bus Error:           false
	Bus Unstacking Fault:               false
	Bus Stacking Fault:                 false
	Bus Lazy FP Fault:                  false
	Undefined Instruction Usage Fault:  false
	Invalid State Usage Fault:          false
	Invalid PC Load Usage Fault:        false
	No Coprocessor Usage Fault:         false
	Unaligned Access Usage Fault:       false
	Divide By Zero:                     false
	Bus Fault on Vector Table Read:     false
	Forced Hard Fault:                  true
	Faulting Memory Address: (valid: false) 0xE000ED34
	Bus Fault Address:       (valid: false) 0xE000ED38
"

---| Fault Status |---
No faults detected.

---| App Status |---
App: signpost_controller   -   [Running]
 Events Queued: 0   Syscall Count: 358   Last Syscall: YIELD

 ╔═══════════╤══════════════════════════════════════════╗
 ║  Address  │ Region Name    Used | Allocated (bytes)  ║
 ╚0x2000E000═╪══════════════════════════════════════════╝
             │ ▼ Grant         408 |    408          
  0x2000DE68 ┼───────────────────────────────────────────
             │ Unused
  0x2000AD58 ┼───────────────────────────────────────────
             │ ▲ Heap         1528 |  14088               S
  0x2000A760 ┼─────────────────────────────────────────── R
             │ Data          10080 |  10080               A
  0x20008000 ┼─────────────────────────────────────────── M
             │ ▼ Stack         328 |   8192          
  0x20007EB8 ┼───────────────────────────────────────────
             │ Unused
  0x20006000 ┴───────────────────────────────────────────
             .....
  0x00040000 ┬─────────────────────────────────────────── F
             │ App Flash     65480                        L
  0x00030038 ┼─────────────────────────────────────────── A
             │ Protected        56                        S
  0x00030000 ┴─────────────────────────────────────────── H

  R0 : 0x00000000    R6 : 0x2000A0B8
  R1 : 0x00000000    R7 : 0x3FB99999
  R2 : 0x00000000    R8 : 0x2000A2F8
  R3 : 0x200089C0    R10: 0x00000009
  R4 : 0x00000001    R11: 0x200089C8
  R5 : 0x00000000    R12: 0x00000000
  R9 : 0x20008000 (Static Base Register)
  SP : 0x20007F40 (Process Stack Pointer)
  LR : 0x00036803
  PC : 0x0003080F
 YPC : 0x00036802

 APSR: N 0 Z 1 C 1 V 0 Q 0
       GE 0 0 0 0
 IPSR: Exception Type - Thread Mode
 EPSR: ICI.IT 0x00
       ThumbBit true 
 To debug, run `make debug RAM_START=0x20006000 FLASH_INIT=0x3005d`
 in the app's folder and open the .lst file.
```

I can consistently fix the problem by commenting out old debugging print statements. None of my segments are at an obvious power of two boundary, which was my first thought. Here is the build size output for a breaking app:

```
   text	   data	    bss	    dec	    hex	filename
  58368	   2420	  20972	  81760	  13f60	build/cortex-m0/cortex-m0.elf
  55676	   2420	  20972	  79068	  134dc	build/cortex-m3/cortex-m3.elf
  55664	   2420	  20972	  79056	  134d0	build/cortex-m4/cortex-m4.elf
```

and here is one for a (barely) working app:

```
   text	   data	    bss	    dec	    hex	filename
  57772	   2388	  20972	  81132	  13cec	build/cortex-m0/cortex-m0.elf
  55076	   2388	  20972	  78436	  13264	build/cortex-m3/cortex-m3.elf
  55064	   2388	  20972	  78424	  13258	build/cortex-m4/cortex-m4.elf
```

I don't have time to debug this right now, and mainly wanted to document it in case anyone else is running into this problem. I'll eventually have to circle back. 

Also, I'm currently running on [this commit](https://github.com/helena-project/tock/commit/e5b92511c91e0f090e16d634f1cb2f36ca40b187), which is a little old. I'm pretty sure that none of the changes since then have addressed this issue. I can try current master in a few weeks.

alevy, 2018-04-03:
How much flash does the SAM4L on signpost have? Cause this particular app is either pushing up against, or exceeding, the 256kB flash address. And it's crashing in the kernel, not in the processes, so could somehow be do to attempting to read an invalid flash address or something.

adkinsjd, 2018-04-03:
We should be running the 512KB version unless our assembler messed up. I
can check the markings on the chip when I get back to lab.

bradjc, 2018-04-05:
I'm experimenting with a "big app" and haven't been able to re-create this problem. Now I'm not really using much of the code in the app, but it is the kernel that is faulting.

App: https://github.com/helena-project/tock/tree/bigapp/userland/examples/tests/bigapp

size:
```
   text	   data	    bss	    dec	    hex	filename
  66936	   1865	  11096	  79897	  13819	build/cortex-m0/cortex-m0.elf
  63832	   1865	  11096	  76793	  12bf9	build/cortex-m3/cortex-m3.elf
  63872	   1865	  11096	  76833	  12c21	build/cortex-m4/cortex-m4.elf
```

kernel debug:

```
Kernel panic at /Users/bradjc/git/tock/kernel/src/process.rs:752:
	"Process crash_dummy had a fault"
	Kernel version release-1.0-2018-02-245-gbc350ab0

---| Fault Status |---
Data Access Violation:              true
Forced Hard Fault:                  true
Faulting Memory Address:            0x00000000
Fault Status Register (CFSR):       0x00000082
Hard Fault Status Register (HFSR):  0x40000000

---| App Status |---
App: bigapp   -   [Yielded]
 Events Queued: 0   Syscall Count: 533   Last Syscall: YIELD

 ╔═══════════╤══════════════════════════════════════════╗
 ║  Address  │ Region Name    Used | Allocated (bytes)  ║
 ╚0x20008000═╪══════════════════════════════════════════╝
             │ ▼ Grant         452 |    452
  0x20007E3C ┼───────────────────────────────────────────
             │ Unused
  0x20006D08 ┼───────────────────────────────────────────
             │ ▲ Heap         1639 |   6043               S
  0x200066A1 ┼─────────────────────────────────────────── R
             │ Data           7841 |   7841               A
  0x20004800 ┼─────────────────────────────────────────── M
             │ ▼ Stack        1560 |   2048
  0x200041E8 ┼───────────────────────────────────────────
             │ Unused
  0x20004000 ┴───────────────────────────────────────────
             .....
  0x00050000 ┬─────────────────────────────────────────── F
             │ App Flash    131028                        L
  0x0003002C ┼─────────────────────────────────────────── A
             │ Protected        44                        S
  0x00030000 ┴─────────────────────────────────────────── H

  R0 : 0x01D1597C    R6 : 0x20004F80
  R1 : 0x20004783    R7 : 0x00000685
  R2 : 0x00000000    R8 : 0x00000704
  R3 : 0x00000000    R10: 0x000006F1
  R4 : 0x00000004    R11: 0x0000075F
  R5 : 0x00000000    R12: 0x00000685
  R9 : 0x20004800 (Static Base Register)
  SP : 0x20004758 (Process Stack Pointer)
  LR : 0x00034BB3
  PC : 0x20004783
 YPC : 0x00034B96

 APSR: N 0 Z 0 C 0 V 0 Q 0
       GE 0 0 1 1
 IPSR: Exception Type - IRQn
 EPSR: ICI.IT 0x12
       ThumbBit false !!ERROR - Cortex M Thumb only!
 To debug, run `make debug RAM_START=0x20004000 FLASH_INIT=0x30051`
 in the app's folder and open the .lst file.

App: crash_dummy   -   [Fault]
 Events Queued: 0   Syscall Count: 6   Last Syscall: YIELD

 ╔═══════════╤══════════════════════════════════════════╗
 ║  Address  │ Region Name    Used | Allocated (bytes)  ║
 ╚0x2000A000═╪══════════════════════════════════════════╝
             │ ▼ Grant         300 |    300
  0x20009ED4 ┼───────────────────────────────────────────
             │ Unused
  0x200091E8 ┼───────────────────────────────────────────
             │ ▲ Heap            0 |   3308               S
  0x200091E8 ┼─────────────────────────────────────────── R
             │ Data            488 |    488               A
  0x20009000 ┼─────────────────────────────────────────── M
             │ ▼ Stack          56 |   4096
  0x20008FC8 ┼───────────────────────────────────────────
             │ Unused
  0x20008000 ┴───────────────────────────────────────────
             .....
  0x00050400 ┬─────────────────────────────────────────── F
             │ App Flash       976                        L
  0x00050030 ┼─────────────────────────────────────────── A
             │ Protected        48                        S
  0x00050000 ┴─────────────────────────────────────────── H

  R0 : 0x00000000    R6 : 0x2000908C
  R1 : 0x00000001    R7 : 0x00000000
  R2 : 0x00000000    R8 : 0x00000000
  R3 : 0x00000000    R10: 0x00000000
  R4 : 0x00000000    R11: 0x00000000
  R5 : 0x20009000    R12: 0x000501B2
  R9 : 0x20009000 (Static Base Register)
  SP : 0x20008FC8 (Process Stack Pointer)
  LR : 0x000500C9
  PC : 0x00050198
 YPC : 0x00050198

 APSR: N 0 Z 1 C 1 V 0 Q 0
       GE 0 0 0 0
 IPSR: Exception Type - Thread Mode
 EPSR: ICI.IT 0x00
       ThumbBit true
 To debug, run `make debug RAM_START=0x20008000 FLASH_INIT=0x50055`
 in the app's folder and open the .lst file.
```

ppannuto, 2018-04-05:
We've had a bunch of fixes around this space lately. I think I'd wait for #859 #857 #854 and #819 all to land before digging into this much more.

bradjc, 2018-05-22:
Any thoughts on how to reproduce this? I just tried my `bigapp` branch which results in an app of this size:

```
   text	   data	    bss	    dec	    hex	filename
  66904	   1916	   5976	  74796	  1242c	build/cortex-m0/cortex-m0.elf
  63800	   1916	   5976	  71692	  1180c	build/cortex-m3/cortex-m3.elf
  63840	   1916	   5976	  71732	  11834	build/cortex-m4/cortex-m4.elf
```

and it seems ok.

alevy, 2018-06-05:
@adkinsjd any chance you can try and either reproduce this or confirm it's been fixed?

bradjc, 2018-06-05:
Closed for now, please re-open if this is still an issue.

closed 2018-06-05 by bradjc

torfmaster, 2018-06-29:
I have similar problem on nrf52dk with the recent (c879d5f7f78d28401f0442e1acb34b13f3d97e74) version of the tock kernel. After running run_hardware_tests.sh:

```
Initialization complete. Entering main loop
NRF52-DK HW INFO: Variant: AAB0, Part: N52832, Package: QF, Ram: K64, Flash: K512
[test-results]
heap_test = "Heap works."
test_ipc = "passed"
test="done"

Kernel panic at /home/philipp/Projects/rust-iot/forks/tock/kernel/src/process.rs:370:
	""
	Kernel version release-1.2-2018-06-158-gc879d5f7

---| Fault Status |---
Instruction Access Violation:       true
Instruction Bus Error:              true
Forced Hard Fault:                  true
Fault Status Register (CFSR):       0x00000101
Hard Fault Status Register (HFSR):  0x40000000

---| App Status |---
App: ble_scanning   -   [Fault]
 Events Queued: 0   Syscall Count: 61   Dropped Callback Count: 0
 Restart Count: 0
 Last Syscall: YIELD

 ╔═══════════╤══════════════════════════════════════════╗
 ║  Address  │ Region Name    Used | Allocated (bytes)  ║
 ╚0x20003000═╪══════════════════════════════════════════╝
             │ ▼ Grant         552 |    552          
  0x20002DD8 ┼───────────────────────────────────────────
             │ Unused
  0x20002080 ┼───────────────────────────────────────────
             │ ▲ Heap       536879232 | 536882648               S
  0x00000000 ┼─────────────────────────────────────────── R
             │ Data              0 |      0               A
  0x00000000 ┼─────────────────────────────────────────── M
             │ ▼ Stack      3758088096 | 3758088192          
  0x20002060 ┼───────────────────────────────────────────
             │ Unused
  0x20002000 ┴───────────────────────────────────────────
             .....
  0x00030000 ┬─────────────────────────────────────────── F
             │ App Flash     65488                        L
  0x00020030 ┼─────────────────────────────────────────── A
             │ Protected        48                        S
  0x00020000 ┴─────────────────────────────────────────── H

  R0 : 0x20002294    R6 : 0x0002BCE8
  R1 : 0x8000BDF4    R7 : 0x0002BCE0
  R2 : 0x00000000    R8 : 0x00000008
  R3 : 0x800088E7    R10: 0x00023803
  R4 : 0x0002BCFC    R11: 0x200021F4
  R5 : 0x0002BD0C    R12: 0x2000220C
  R9 : 0x20002400 (Static Base Register)
  SP : 0x200021C8 (Process Stack Pointer)
  LR : 0x00023683
  PC : 0x800088E6
 YPC : 0x00028A04

 APSR: N 0 Z 0 C 1 V 0 Q 0
       GE 0 0 0 0
 EPSR: ICI.IT 0x00
       ThumbBit true 
 To debug, run `make debug RAM_START=0x20002000 FLASH_INIT=0x20059`
 in the app's folder and open the .lst file.

App: hardware_test   -   [Yielded]
 Events Queued: 0   Syscall Count: 63   Dropped Callback Count: 0
[... 66 lines cut ...]
             │ App Flash      4040                        L
  0x00032038 ┼─────────────────────────────────────────── A
             │ Protected        56                        S
  0x00032000 ┴─────────────────────────────────────────── H

  R0 : 0x00032AED    R6 : 0x00000000
  R1 : 0x00000000    R7 : 0x00000000
  R2 : 0x00000000    R8 : 0x00000000
  R3 : 0x00032075    R10: 0x00000000
  R4 : 0x20004400    R11: 0x00000000
  R5 : 0x00000000    R12: 0x20004410
  R9 : 0x20004400 (Static Base Register)
  SP : 0x200043F0 (Process Stack Pointer)
  LR : 0x000003F0
  PC : 0x00000000
 YPC : 0x00032AEC

 APSR: N 0 Z 0 C 1 V 0 Q 0
       GE 0 0 0 0
 EPSR: ICI.IT 0x11
       ThumbBit false !!ERROR - Cortex M Thumb only!
 To debug, run `make debug RAM_START=0x20004000 FLASH_INIT=0x32061`
 in the app's folder and open the .lst file.

```

bradjc, 2018-06-29:
Do you know of a tock commit where that app works?

I am trying this on a nRf52840dk with a kernel from the Tock commit you mentioned. What I am seeing is that https://github.com/tock/libtock-rs/blob/master/src/simple_ble.rs#L20 should be `0x01`, and that the kernel when running `ble_scanning.rs` is panicing. The app itself is also panicing (causing all of the leds to blink). The app panic seems to be coming from the unwrap on this line:

https://github.com/tock/libtock-rs/blob/812297dde9ae2ec8edb8be8deaaa63d41c02f075/examples/ble_scanning.rs#L36

I'm not really sure what is causing the app to actually get into a fault state that causes the kernel to panic.

I also tried the c version of the ble scanning app and that seems to run.

bradjc, 2018-06-29:
And testing with this commit https://github.com/tock/libtock-rs/pull/53 it seems that the stack is overflowing. Note I have commented out most of the code in the ble_scanning.rs app so that it doesn't panic on an unwrap.

```
Kernel panic at /Users/bradjc/git/tock/kernel/src/process.rs:370:
	""
	Kernel version release-1.2-2018-06-158-gc879d5f7

---| Fault Status |---
Memory Management Stacking Fault:   true
Forced Hard Fault:                  true
Fault Status Register (CFSR):       0x00000010
Hard Fault Status Register (HFSR):  0x40000000

---| App Status |---
App: ble_scanning   -   [Fault]
 Events Queued: 0   Syscall Count: 6246   Dropped Callback Count: 0
 Restart Count: 0
 Last Syscall: COMMAND

 ╔═══════════╤══════════════════════════════════════════╗
 ║  Address  │ Region Name    Used | Allocated (bytes)  ║
 ╚0x20006000═╪══════════════════════════════════════════╝
             │ ▼ Grant         648 |    648
  0x20005D78 ┼───────────────────────────────────────────
             │ Unused
  0x20004000 ┼───────────────────────────────────────────
             │ ▲ Heap            0 |   7544               S
  0x20004000 ┼─────────────────────────────────────────── R
             │ Data           2048 |   2048               A
  0x20003800 ┼─────────────────────────────────────────── M
             │ ▼ Stack        6152 |   6144 EXCEEDED!
  0x20001FF8 ┼───────────────────────────────────────────
             │ Unused
  0x20002000 ┴───────────────────────────────────────────
             .....
  0x00031000 ┬─────────────────────────────────────────── F
             │ App Flash      4048                        L
  0x00030030 ┼─────────────────────────────────────────── A
             │ Protected        48                        S
  0x00030000 ┴─────────────────────────────────────────── H

  R0 : 0x20002B48    R6 : 0x00000012
  R1 : 0x20002B94    R7 : 0x200037C3
  R2 : 0x20002BDC    R8 : 0x2000379C
  R3 : 0x00000424    R10: 0x00000012
  R4 : 0x200037AB    R11: 0x00000013
  R5 : 0x200037AD    R12: 0x20002808
  R9 : 0x20003800 (Static Base Register)
  SP : 0x20001FF8 (Process Stack Pointer)
  LR : 0x00000001
  PC : 0x0003009C
 YPC : 0x00030112

 APSR: N 0 Z 1 C 1 V 0 Q 0
       GE 0 0 0 0
 EPSR: ICI.IT 0x00
       ThumbBit true
 To debug, run `make debug RAM_START=0x20002000 FLASH_INIT=0x30059`
 in the app's folder and open the .lst file.
```