tock/tock #866: Replace `make` with `cargo`

niklasad1, 2018-04-02:
[thumb* support in Cargo](https://github.com/rust-lang/rust/pull/49563) is on its way which probably will help

I think it should be possible using different [workspaces](https://doc.rust-lang.org/book/second-edition/ch14-03-cargo-workspaces.html) for the different boards with separate flashing. One nice thing is that all builds will be in the same output folder

bradjc, 2018-04-05:
That was just merged. Does that mean we can get rid of xargo??

ppannuto, 2018-04-05:
I think we'll need at least this as well: https://github.com/rust-lang/rust/issues/49380 :/

whitequark, 2018-04-07:
That was just merged too.

Woyten, 2018-04-08:
I already played around with Cargo's thumb* support on `libtock-rs`. Currently, I think that the quality is not good enough yet:
- https://travis-ci.org/torfmaster/libtock-rs/builds/363729498
- https://travis-ci.org/torfmaster/libtock-rs/builds/363724011

bradjc, 2018-04-08:
I have a branch https://github.com/helena-project/tock/tree/update-nightly-2018-04-08 that updates Tock to use the nightly with those patches. However, it looks like we won't be able to actually update until rustfmt pushes a new version because some rustfmt dependencies won't compile with the new nightly.

bradjc, 2018-04-15:
It looks like we really need this: https://github.com/rust-lang/cargo/issues/4959

bradjc, 2018-04-30:
>To remove the need for arm-none-eabi we need: rust-lang/rust#49584

Which is now a PR: https://github.com/rust-lang/rust/pull/50336

bradjc, 2018-06-22:
> Which is now a PR: rust-lang/rust#50336

That got merged, but then broke rust nightly for everyone. Now we are waiting on this fix: https://github.com/rust-lang/rust/pull/51703

bradjc, 2018-06-23:
Ok fix is applied and nightlies work again. However, adding the llvm tools component doesn't seem to work.

https://github.com/rust-lang/rust/issues/49584#issuecomment-399623679

bradjc, 2018-06-25:
We are now waiting on the very unreliable bors/travis to actually succeed

https://github.com/rust-lang/rust/pull/51728

bradjc, 2018-06-25:
rust-lang/rust#51728 was just merged! So with tonight's nightly we should no longer need arm-none-eabi-gcc.

bradjc, 2018-07-06:
Eh, the more I think about using cargo-binutils I don't think it is worth it. It is effectively a wrapper around the same thing we are doing now, and _adds_ a new dependency that isn't nicely version tracked as a rustup component.

closed 2018-07-06 by bradjc