tock/tock #874: [gpio hil/capsule] add functionlity to specify `gpio drive configuration`

niklasad1, 2018-04-09:
https://github.com/helena-project/tock/pull/867#discussion_r130560976

fhars, 2018-04-14:
I am not quite sure that this is actually consistent enough over all possible chips tock might run on to make a general interface at the HIL level possible. You usually need this functionality if you are very close to the actual hardware (I needed it to implement I2C on the NRF51) where you already have to know exactly which chip you are dealing with. If this isn't designed very carefully you might end up with something that will just slightly miss what you actually need so you have to bypass it anyway to get anything done (the constant bane of all hardware abstraction layers).

niklasad1, 2018-04-14:
It is probably not consistent between chips but my idea was to have an enum for each gpio or something like:

enum PinDriveConfiguration {
   Low,
   Mid,
   High,
}

Then do similar pattern matching as in https://github.com/helena-project/tock/blob/master/chips/nrf5x/src/gpio.rs#L348-#L356

Should be trivial to change it with: `gpio_regs.pin_cnf[self.pin as usize].write(pin_config + PinCfg::DRIVE::H0H1);` for example for high ´0´ and ´1´ 

Otherwise I guess that you have configure it at compile-time (which may be ok as well) or implement a special `nrf5x libtock` library for that!

fhars, 2018-04-14:
But what would be the enum that would abstract over both of the NRF51 set of options
Standard '0', standard '1'
High drive '0', standard '1'
Standard '0', high drive '1'
High drive '0', high 'drive '1''
Disconnect '0' standard '1'
Disconnect '0', high drive '1'
Standard '0'. disconnect '1'
High drive '0', disconnect '1'

and the TI CC26x0 one of
2mA
4mA
8mA (for some pins)
automatically based on battery monitor,
all either normal, open source or open drain?

niklasad1, 2018-04-15:
Haha, good point I don't know really!

I guess we have to treat this as `chip-specific` either by a user-space library or just determine it at compile-time then!

closed 2018-04-15 by niklasad1