tock/tock #875: [register interface] why is individual fields interpretted as isize

niklasad1, 2018-04-09:
Take this example:

```rust
register_bitfields! [u32,
    InfoPart [
        PART OFFSET(0) NUMBITS(32) [
            /// nRF52838
            N52832 = 0x52832,
            /// Unspecified
            Unspecified = 0xffffffffu32
        ]
    ]
];
```
```bash
error[E0308]: mismatched types             
   --> /home/niklasad1/GitHub/tock/chips/nrf52/src/ficr.rs:130:27                     
    |                                      
130 |             Unspecified = 0xffffffffu32                                         
    |                           ^^^^^^^^^^^^^ expected isize, found u32               

error: aborting due to previous error 
```

bradjc, 2018-04-09:
I think that is the default in rust? I noticed this to, and tried setting it here:

https://github.com/helena-project/tock/blob/0188f7a980a25473f7d5adfd8be26f8432182f8a/kernel/src/common/regs/macros.rs#L90-L98

Which didn't work because rust wasn't happy with the register definitions with no contained values.

niklasad1, 2018-04-10:
Yeah, you are right @bradjc the default values are signed integers

It is not possible to specify the representation for a `zero-variant enum`, https://doc.rust-lang.org/error-index.html#E0084

closed 2018-04-10 by niklasad1