tock/tock #889: ble_passive_scanning not building

dcellucci, 2018-04-18:
## The problem
In the `ble_passive_scanning` example in the userland, I get this error when I run `make flash` on a NRF52832 (with the proper TOCK_BOARD flag) 

```
advertisement.cpp:66:3: warning: identifier 'nullptr' is a keyword in C++11 [-Wc++0x-compat]
   if (buf != nullptr && len >= ADDRESS_SIZE && buf[0] <= 6) {
   ^
advertisement.cpp: In static member function 'static bool Advertisement::validAdvertisement(const unsigned char*, int)':
advertisement.cpp:66:14: error: 'nullptr' was not declared in this scope
   if (buf != nullptr && len >= ADDRESS_SIZE && buf[0] <= 6) {
              ^
advertisement.cpp:71:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
```

## The Solution
I fixed this problem by adding a flag `-std=gnu++0x\` to the CPPFLAGS in `userland/Configuration.mk`. (Line 45).

It's a single line so I'm not sure if it deserves the full pull request, but if you'd rather I do it that way, I can.

alevy, 2018-04-18:
@dcellucci what version of arm-none-eabi-g++ are you using? It compiles fine for me as well as on the CI, so I'm __guessing__ it's a version mismatch problem.

(Having said that, I'm not opposed to just adding a C++11 constraint if that helps support more versions of gcc)

dcellucci, 2018-04-18:
oh boy- 
`arm-none-eabi-g++ --version` gives:
```
gcc version 5.2.1 20151202 (release) [ARM/embedded-5-branch revision 231848] (GNU Tools for ARM Embedded Processors)
```
It works after I update to the latest version... sorry about that!

closed 2018-04-18 by alevy