tock/tock #897: `JLinkGdbServer` renamed  to `JLinkGDBServerCLExe` in the newest version

niklasad1, 2018-04-21:
```
$ JLinkExe
J-Link>f
Firmware: J-Link OB-SAM3U128-V2-NordicSemi compiled Jan 12 2018 16:05:20               
Hardware: V1.00
```

```
$ JLinkGDBServer
zsh: command not found: JLinkGDBServer
```
```
$ JLinkGDBServerCLExe
│SEGGER J-Link GDB Server V6.30k Command Line Version                                                                                       
JLinkARM.dll V6.30k (DLL compiled Apr  9 2018 18:33:07)
```

Anyone else who has this problem?

ppannuto, 2018-04-24:
Ay yi yi, that's a bit obnoxious. Should be able to update the scripts to check which exists and use the right one:

     JLinkGdb=$(hash JLinkGDBServer2 2> /dev/null && echo JLinkGDBServer || echo JLinkGDBServerCLExe)
     $JLinkGdb --args ...

brghena, 2018-04-26:
What method did you use to install the JLink tools?

I just installed v6.32 from the DEB in Ubuntu, and it created (among other commands):

```
lrwxrwxrwx 1 root root 42 Apr 20 08:35 /usr/bin/JLinkGDBServer -> /opt/SEGGER/JLink_V632/JLinkGDBServerCLExe
lrwxrwxrwx 1 root root 42 Apr 20 08:35 /usr/bin/JLinkGDBServerCLExe -> /opt/SEGGER/JLink_V632/JLinkGDBServerCLExe
lrwxrwxrwx 1 root root 40 Apr 20 08:35 /usr/bin/JLinkGDBServerExe -> /opt/SEGGER/JLink_V632/JLinkGDBServerExe
```

So it did rename the GDB server, but the original name still works too.

alevy, 2018-04-26:
Out of left field maybe, but what about switching wholesale to OpenOCD? it's easier to obtain anyway on most platforms: on most Linux distros it's in the package manager and I assume it's in Homebrew as well.

niklasad1, 2018-04-26:
Ok, installed via AUR on arch linux! 

If, nobody else has this problem I’d be happy to create a symlink and close this :)

bradjc, 2018-04-26:
> what about switching wholesale to OpenOCD

Feels like boards should support whatever they want to, but we could push openocd to be at least an option for each board. It needs to be supported by tockloader, however, before it's useful for me at least.

https://github.com/tock/tockloader/issues/19

bradjc, 2018-05-22:
If anyone runs into this problem please re-open.

closed 2018-05-22 by bradjc