tock/tock #905: Alarm Capsule should do a check in a loop

bradjc, 2018-04-25:
[image]

bradjc, 2018-06-06:
I think this is actually OK as is.

https://github.com/tock/tock/blob/0ba03dd809e3b41ac5d28d816d95c95eee2872c2/capsules/src/alarm.rs#L184-L189

calls `reset_active_alarm` which, as implied, provides the next alarm sorted by expiration time. 

https://github.com/tock/tock/blob/0ba03dd809e3b41ac5d28d816d95c95eee2872c2/capsules/src/alarm.rs#L48-L71

Therefore if that alarm _hasn't_ expired, then no other alarms have expired, either. If that alarm _has_ expired, then it recursively calls the `fired()` method, which will run the same check again, and check the next timer (if it exists) and if that one has fired as well, then it will recurse as well. I believe the recursion handles the loop.

Does that make sense?

brghena, 2018-06-07:
Oh, yeah, that does make sense to me. I hadn't realized the recursive `fired()` call was in there.

closed 2018-06-07 by brghena