tock/tock #907: SysTick::value math is wrong

phil-levis-google, 2018-04-25:
https://github.com/tock/tock/blob/97f9a35770d47db8e58aa411ceb82b7fcafb1771/arch/cortex-m4/src/systick.rs#L111

This should be 

(value * 1_000_000) / hertz

not

(value / hertz)  / 1_000_000

which is the same as

value / (hertz * 1_000_000)

Changing to the fixed line safely requires using 64-bit values. Generally speaking, the software for value and set_timer should use intermediate 64-bit values to avoid the arithmetic roundoff acrobatics.

alevy, 2018-04-26:
This is a fairly easy fix, with one caveat: it costs about 0.6kB of code (and the associated run-time overhead) because long division on ARM is a fairly expensive subroutine.

In _practice_ this overhead is avoided for `set_timer` because the compiler inlines it, and the value we pass in is a large enough power of 10 that the compiler can figure out to simplify it with the `1_000_000` divisor and stick to normal 32-bit division.

However, we don't get that luxury in `value()`, which is called on every context switch to determine if there is enough time in the quantum left to be worth context switching to the process.

One possibility would be to just modify the interface slightly. `value()` is only used to then compare against a threshold value in the process scheduler. We could, instead, expose a `greater_than(&self, us: u32)` method that performed the comparison, but lends itself to the same compiler optimization when the threshold value is round (currently it's `500` in the scheduler).

Doing this actually ends up _saving_ a handful of bytes.

Thoughts?

phil-levis-google, 2018-05-02:
Sorry this went to my Google account, which I only check on Wednesdays.

I'd rather avoid structuring software in a way that assumes/is optimized the values we're currently calling the functions with (the "in practice" part). I.e., it's great if in practice the cost isn't there but let's be OK if the cost is there.

One other approach that does some gymnastics based on cases:

if value < 1000 {
    (value * 1_000_000) / hertz 
} else if value > 1_000_000 {
    (value / 1000) * (1_000_000_000 / hertz)
} else { // 1000 <= value <= 1_000_000
    ((val * 1000) / hertz) * 1000
}

alevy, 2018-05-03:
@phil-levis I can't tell if you're for or against the solution in #910. Can you please approve/comment/request changes there?

PR #910 (2018-05-08): Improve systick precision
### Pull Request Overview

Use 64-bit arithmetic in the systick implementations to convert from
microseconds to tics.

64-bit division is a _very_ expensive operation in
Thumb, but as long as the microsecond value is a reasonably
large multiple of 10, the conversion is optimized by the compiler to
simple 32-bit division/multiplication.

To facilitate this optimization, the PR also changes the systick
interface by replacing the `value` method with a `greater_than` method,
since we only ever use the value in order to do a comparison with a
relatively large multiple of 1 [... cut]
commit 46ef6595 (2018-05-08): Improve systick precision

Uses 64-bit arithmetic in the systick implementations to convert from
microseconds to tics. 64-bit division is a _very_ expensive operation in
Thumb, but as long as the microsecond value is a reasonably
large multiple of 10, the conversion is optimized by the compiler to
simple 32-bit division/multiplication.

To facilitate this optimization, the commit changes the systick
interface by replacing the `value` method with a `greater_than` method,
since we only ever use the value in order to do a comparison with a
relatively large multiple of 10 (500).

Closes #907
closed 2018-05-08 by ppannuto