tock/tock #912: 6lowpan Test Transmission Bug

jlwatson, 2018-04-30:
Hi everyone,

While taking a crack at the hardcoded delay loops in the `ipv6_lowpan_test` code (discussed in #617), I've noticed that every so often (approximately once per test run, ~1/84 packets) the RF233 driver reports a successful transmission but neither the packet sniffer nor the receiver receives it and the overall test hangs. I'm still working on a fix but I wanted to give a heads-up as the 6loWPAN merge goes in.

#### Testing code

My working setup is pushed to Paul's fork [here](https://github.com/ptcrews/tock/tree/sixlowpan_library_test_delay) -- it essentially looks like this on two imixes, one receiving, one transmitting.

```
+------------------------------+                   +------------------------------+
|                              |                   |                              |
|         Transmitter          |                   |          Receiver            |
| MAC: 10:11:12:13:14:15:16:17 | +---------------> | MAC: 20:21:22:23:24:25:26:27 |
|         PAN: 0x0000          |                   |         PAN: 0x0000          |
|                              |                   |                              |
+------------------------------+                   +------------------------------+
```

So far, I've narrowed the issue down to somewhere in the RF233 state machine, but I haven't managed to get to it yet because inserting `debug!` statements that close changes the timing and the tests pass. If anyone has any thoughts, let me know!

ppannuto, 2018-05-01:
As a testing aid, you may try using `debug_gpio!` instead of printing debug. Attach a saleae for timestamping the long-running trace and to let you monitor a bunch of gpios.

jlwatson, 2018-05-01:
Forgot about that -- thanks!

PR #924 (2018-05-08): Detect RF233 CHANNEL_ACCESS_FAILURE (max CSMA retry) failures
### Pull Request Overview

Fixes #912. Transmit clients would receive `ReturnCode::SUCCESS` in `send_done` even after a transmission failed by exceeding the maximum number of CSMA retries. The `rf233` driver now returns `ReturnCode::FAIL` in this case; if detected, clients should attempt resend the packet.

### Testing Strategy

@phil-levis found the issue in SPI output; tested expected behavior using setup from #912.

### Documentation Updated

- [X] Kernel: Updated the relevant files in `/docs`, or no updates are required.
- [X] U [... cut]
closed 2018-05-08 by bradjc