tock/tock #930: Create a crate for Register/Bitfield Interface?

andre-richter, 2018-05-11:
Hi guys,

I read your recent blog post about your register/bitfield interface. For me it was a perfect timing to learn about it, since we recently started a discussion about register R/W semantics in the Rust embedded-wg: https://github.com/rust-lang-nursery/embedded-wg/issues/87

One thing we were missing so far was a good common approach towards bitfields (bitflags was kind of straight forward through the bitflags crate). It seems your code has, among other useful functionality, a neat solution to this already.
Hence I am asking you if you would be open to engage in that discussion, and in that context would be open to carving out a (standalone) crate of your current code? This would make it more accessible to others.

In the discussion linked above, we also concluded that these register operations are not only applicable for MMIO, but to some extent also for CPU registers. Hence, the code could be reused even more by abstracting the reading/writing of the raw register bits according to the actual source (MMIO, CPU; Maybe introduce interface traits for it?), and share the modification functions. Another minor thing I noticed from skimming through your code is lack of u64 support, but that shouldn't be a big issue ofc.

Kindly let me know what you think about it.

alevy, 2018-05-11:
My thoughts:

> Hence I am asking you if you would be open to engage in that discussion

Of course! how?

> would be open to carving out a (standalone) crate of your current code

Almost certainly. We could (and should) definitely move that code into a separate crate in the repo, and potentially publish it on crates.io. Eventually, if other people use it, it probably makes sense to pull it out to a separate repo, but for now we're still finding that it's useful to be able to develop these various intertwined libraries for Tock in tandem with in the same repo. This just means that getting pull requests for such a crate would be somewhat cumbersome at first.

bradjc, 2018-05-11:
Where should we put that crate? I don't think it can go inside of the kernel crate if it was a crate itself.

niklasad1, 2018-05-11:
Can't we just create a new crate `tock-register-interface` and place it in tock/tools? Ofc, it would require some refactoring!

Then, it would be beneficial to use workspaces to use straight-forward imports using `git path` to import the crate outside Tock in Cargo.toml!

brghena, 2018-05-11:
I played around with this a bit today to see what it would take. Results are in the `seperate-registers` branch (https://github.com/tock/tock/compare/seperate-registers) which does compile for all boards and the code is tested on Hail.

Currently, we only use the Tock registers interface in the `chips/` and `arch/` directories. The only real changes they need are to their `Cargo.toml` and `src/lib.rs` files in order to make the proper connections. The registers code itself has no reliance on the rest of Tock (which is good) and so can easily be pulled into its own crate with no dependencies (it uses things in `Core`, but no crates).

This branch only pulls it to a separate directory in the Tock repo, but I think moving forward the best choice would be to move all the Registers stuff to its own repo, which could then be pushed to crates.io.

The registers interface does use `unsafe`. Possibly we may not want to allow `chips/` or `arch/` to depend on crates which use `unsafe`, even if this is a vetted crate. Is there a way for the kernel to depend on the registers crate and then re-export it to the rest of the system?

Open questions:
- Should the registers interface move to a separate repo or stay in Tock?
- Does having the registers crate outside of the kernel cause problems logically?
- What should the registers interface be called?

andre-richter, 2018-05-12:
Thanks for the effort everyone.

I quickly hacked together the CPU register interface I was talking about earlier, based on @brghena 's branch. You can find the adaptions on the tock side here: https://github.com/andre-richter/tock/commit/fff0fbf3627c18b801c9aa273a19eaeceab0aea3.

I put a small demo of CPU and MMIO registers in action into one of my raspi3 mini kernels here: https://github.com/andre-richter/rust-raspi3-tutorial/commit/7d0a7e6ce77863572c8bf3d4bc0823f54bf5f0f4.

Obviously, this is only a PoC, there's too much boilerplate/duplicate code. I am also not 100% convinced about the function pointer approach, since it introduces a function call overhead. But it was the most idiomatic way I could quickly come up with given the current tock-register-interface design, and making the register interface agnostic of the actual CPU assembly instructions to get and set the register contents.

andre-richter, 2018-05-12:
Introducing traits to reduce boilerplate: https://github.com/andre-richter/tock/commit/40cad0cac20f2b17009ca96521ca60826ac4284a

Stopping here for now, in case you're not interested in this whole CPU thingy? :sweat_smile:

brghena, 2018-05-14:
Okay @andre-richter thanks for the links. I finally understand what you're doing. Using the register system to make registers accessed through `msr`/`mrs` look just like MMIO with accessible fields is really neat. I like the interface you showed in `cntp_ctl_el0.rs` a lot.

However, from a Tock perspective, I don't think it's something we need. Cortex-Ms have very few registers that need to be accessed in that way. And the only time we do so is in the context-switching assembly, which doesn't need this Rust interface.

So, I _think_ what we should do on our end is move forward with making a crate out of the registers stuff. Then, I believe you would still be able to use it to build what you want. The `register_bitfields!` macro you could use directly, and the `RegisterReadWrite` and etc. traits can be made in a separate file by using the `RegisterLongName`, `IntLike`, and `Field` interfaces which we publicly export. Will that work for you @andre-richter , or is there a problem with that which I am missing?

andre-richter, 2018-05-14:
Yes, my intention is to _not_ differentiate between MMIO and CPU registers from an interface perspective, because in the end, both are registers, just the access method differs (MMIO, mrs/msr, mov, you name it).

My overall goal is to enable people to write "rich kernels" in Rust for more complex architectures, like x86 or Cortex-A, that could leverage the same register interface for e.g. device drivers accessed through MMIO, as well as for "coprocessor" drivers (e.g. MMU or crypto stuff) that operate mostly via CPU registers.
IMO this makes sense, and would really improve readability and streamline the code. In current projects, I couldn't find such a unified approach. Everybody does it differently. 

I would start using it in my `cortex-a` crate, and my raspi3 tutorials where I have MMIO as well, and try to act as a good example hopefully inspiring others :)

By inspection, I am not sure how feasible it is to "monkey-patch" onto the current design what I presented in comments above. I guess I would have to try and see if the result is acceptable.
Other ways I could think of is using feature flags to make stuff modular (like MMIO as default, CPU as opt-in). Or as a last resort a fork, but this is my least favorite option because I prefer to leave credit where credit is due 👍

andre-richter, 2018-05-15:
> So, I think what we should do on our end is move forward with making a crate out of the registers stuff. Then, I believe you would still be able to use it to build what you want. The [...]

Just to double check, the separate file for the R/W traits would be part of your delivery or not?
If yes, then I think your suggestion could work fine. CPU crates like `cortex-a` could then implement these traits for their registers, and reuse `register_bitfields!`, `RegisterLongName` `IntLike` `Field` from your crate for that purpose.

andre-richter, 2018-05-31:
Hi,

just wanted to check in again here about the status. Do you have a timeline maybe?

Thanks and sorry for bugging!

brghena, 2018-06-04:
Andre, thanks for poking on this issue. I've gotten distracted with other work going on.

I will get the PR ready that makes a separate crate out of this later this week (probably Wednesday or Thursday). I'm pretty sure this will suit both of our needs, and I'll write up some example code that uses it like you plan to to ensure that.

brghena, 2018-06-07:
Okay, I combined Andre's upgrades with Niklas's example repo. Here: https://github.com/brghena/tock-register-example

@andre-richter I think this demonstrates that you can do everything you want to extern to Tock, using the tock register system as an external crate. Take a look when you get a chance and let me know if you see any problems.

To make this work, I did have to make some fields within register structs public so they can be modified. I don't know if there was a better way to handle that. https://github.com/tock/tock/commit/8fe70726231f7987178144a94b66e82449453b04

PR #984 (2018-06-21): Create crate for Tock Registers
### Pull Request Overview

This PR creates a separate crate for the Tock register interface (named `tock-regs`) and moves the code to `tools/tock-register-interface/`. While this change affects many files throughout Tock, almost all of them are just changes to the import statement. Additionally, `Cargo.toml` files have changed to reference the new crate, and documentation has been updated.

The driver for this change is to allow other projects to use or extend Tock registers. In order to support this, the `mask` and `shift` fields of `Field` as well as the `mask [... cut]
closed 2018-06-21 by bradjc