tock/tock #951: DMA interrupt issue for SPI

hchiang, 2018-05-23:
If you try to send a lot of data over SPI, it only succeeds when the clock is divided by at least ~140.

The SPI capsule uses buffers of size 64, so if you try to send 200 bytes of data, it gets divided into four calls to `read_write_bytes`. The problem is that if the clock used is not that much faster than the SPI's baud rate (the clock divider value SCBR is set to less than ~140), only the first (sometimes also second) call to `read_write_bytes` succeeds. 

It looks like it might be a DMA issue, since SPI's `xfer_done` function expects two interrupts for `read_write_bytes`, one for SPI_TX followed shortly by another for SPI_RX. Instead, when the clock divisor value is small, only the SPI_TX interrupt is received and the SPI hangs waiting for the SPI_RX interrupt. What's odd about this is that it doesn't always fail after the first `read_write_bytes`; in some cases the SPI_RX interrupt is received for the first `read_write_bytes` but is not received for the second call to `read_write_bytes`.

It looks like an interrupt timing issue, but I have not been able to track down the cause and help would be greatly appreciated.

If the SPI sets SCBR to be about 140 or greater, you get the following:
[image]

If the SPI sets SCBR to be less than 140, you get one of the following:
[image]
[image]

bradjc, 2018-05-24:
Is it reasonable to just enforce that the clock has to be fast enough, given the SPI clock speed?

I've spent way too much time in the black hole that is the SAM4L, and by all accounts the SAM4L does some weird stuff with clocks.

hchiang, 2018-05-24:
It kind of makes it hard for the SPI to have a baud rate above ~300Kbit/s and it's possible it will become an issue with other peripherals if they try using some of the slower clocks like the RC1M.

phil-levis, 2018-05-24:
Are SPI_RX and SPI_TX GPIO lines you are toggling in the interrupts? How are you generating those signals?

hchiang, 2018-05-24:
They're gpio_debug outputs.

phil-levis, 2018-05-24:
When exactly do they occur? I mean, the delay between the SPI completion and the TX in the lower figure is huge...

hchiang, 2018-05-24:
That figure was generated using the RC1M as the system clock whereas the above graphs were generated using DFLL or RCFAST.

phil-levis, 2018-05-24:
In what exact piece of code are the pins toggled?

hchiang, 2018-05-24:
At the beginning of the xfer_done function of spi.rs, though the output is almost exactly the same if you instead toggle them in the handle_interrupt function of dma.rs.

hchiang, 2018-05-26:
It seems to be an issue with the DMA and not with SPI since I observed a similar issue for the ADC. When running the userland/examples/tests/adc application with the main clock set to RC1M, the interrupt is not triggered for buffered ADC samples at >= 5000 Hz. The application works as expected for other clocks including RCFAST at 4MHz sampling at a rate of 175 KHz.

alevy, 2018-05-26:
This seems _most likely_ to be a logic error in the DMA driver that happens not to manifest with slower clocks.

closed 2018-05-29 by hchiang